ccMSC/ckb #10: K65 support?

richlewis42, 2015-01-07:
How straightforward would adding support for the tenkeyless K65 model?  I've looked through the code a bit but not too familiar with C.  The output of lsusb if handy:

$ lsusb -vd 1b1c
USB:

    USB 3.0 SuperSpeed Bus:

      Host Controller Location: Built-in USB
      Host Controller Driver: AppleUSBXHCI
      PCI Device ID: 0x1e31 
      PCI Revision ID: 0x0004 
      PCI Vendor ID: 0x8086 
      Bus Number: 0x0a 

    USB Hi-Speed Bus:

      Host Controller Location: Built-in USB
      Host Controller Driver: AppleUSBEHCI
      PCI Device ID: 0x1e26 
      PCI Revision ID: 0x0004 
      PCI Vendor ID: 0x8086 
      Bus Number: 0x1d 

        Hub:

          Product ID: 0x0024
          Vendor ID: 0x8087  (Intel Corporation)
          Version: 0.00
          Speed: Up to 480 Mb/sec
          Location ID: 0x1d100000 / 2
          Current Available (mA): 500
          Current Required (mA): 0
          Built-In: Yes

            Hub:

              Product ID: 0x2512
              Vendor ID: 0x0424  (SMSC)
              Version: b.b3
              Speed: Up to 480 Mb/sec
              Location ID: 0x1d180000 / 3
              Current Available (mA): 500
              Current Required (mA): 2
              Built-In: Yes

                BRCM20702 Hub:

                  Product ID: 0x4500
                  Vendor ID: 0x0a5c  (Broadcom Corp.)
                  Version: 1.00
                  Speed: Up to 12 Mb/sec
                  Manufacturer: Apple Inc.
                  Location ID: 0x1d181000 / 5
                  Current Available (mA): 500
                  Current Required (mA): 94
                  Built-In: Yes

                    Bluetooth USB Host Controller:

                      Product ID: 0x8286
                      Vendor ID: 0x05ac  (Apple Inc.)
                      Version: 1.47
                      Speed: Up to 12 Mb/sec
                      Manufacturer: Apple Inc.
                      Location ID: 0x1d181300 / 6
                      Current Available (mA): 500
                      Current Required (mA): 0
                      Built-In: Yes

                Apple Internal Keyboard / Trackpad:

                  Product ID: 0x0263
                  Vendor ID: 0x05ac  (Apple Inc.)
                  Version: 2.22
                  Speed: Up to 12 Mb/sec
                  Manufacturer: Apple Inc.
                  Location ID: 0x1d182000 / 4
                  Current Available (mA): 500
                  Current Required (mA): 40
                  Built-In: Yes

    USB Hi-Speed Bus:

      Host Controller Location: Built-in USB
      Host Controller Driver: AppleUSBEHCI
      PCI Device ID: 0x1e2d 
      PCI Revision ID: 0x0004 
      PCI Vendor ID: 0x8086 
      Bus Number: 0x1a 

        Hub:

          Product ID: 0x0024
          Vendor ID: 0x8087  (Intel Corporation)
          Version: 0.00
          Speed: Up to 480 Mb/sec
          Location ID: 0x1a100000 / 2
          Current Available (mA): 500
          Current Required (mA): 0
          Built-In: Yes

            FaceTime HD Camera (Built-in):

              Product ID: 0x8510
              Vendor ID: 0x05ac  (Apple Inc.)
              Version: 80.25
              Serial Number: CC2D9G053DDN9KB0
              Speed: Up to 480 Mb/sec
              Manufacturer: Apple Inc.
              Location ID: 0x1a110000 / 3
              Current Available (mA): 500
              Current Required (mA): 500
              Built-In: Yes

    USB 3.0 Hi-Speed Bus:

      Host Controller Location: Built-in USB
      Host Controller Driver: AppleUSBXHCI
      PCI Device ID: 0x1e31 
      PCI Revision ID: 0x0004 
      PCI Vendor ID: 0x8086 
      Bus Number: 0x0a 

        Corsair K65 RGB Gaming Keyboard:

          Product ID: 0x1b17
          Vendor ID: 0x1b1c  (CORSAIR MEMORY INC.)
          Version: 1.13
          Serial Number: 0203A03DAE3998A7534EDC40F5001940
          Speed: Up to 12 Mb/sec
          Manufacturer: Corsair
          Location ID: 0x14200000 / 13
          Current Available (mA): 500
          Current Required (mA): 500

ccMSC, 2015-01-08:
So, I merged your pull request, but I was also working on some daemon code at the same time. Test the latest commit for me and let me know if it's still working.

The ckb app basically sees it as a K70, right? Should be pretty easy to get the K65 layout implemented there.

richlewis42, 2015-01-10:
It's my office keyboard for the moment, so I'll trial the new commit on Monday - thanks again!  

Yep I noticed that the pinwheel animation isn't centered on the K65, and the media buttons are out of sync, so it must think it's a K70.  I checked out the keymap source code, I could possibly submit another pull request adding the K65 layout (just removing numpad keys I think) at some point, but I'm busy for the next few days, and it looks much more involved, so I'll have to see.

ccMSC, 2015-01-10:
Ok, fair enough. I think it would be fairly easy to construct a K65 map, you could basically just take the K70 map and chop off anything with an X value >= the NumLock key. You'd have to make it reposition the media keys manually, though.

By the way, can you find out which of the K70 keys control the Volume Up/Volume Down buttons? Or is ckb not able to change them at all? I would presume the Mute and Win Lock keys have the same codes as they do on the K70, but since the K70's volume wheel doesn't have LEDs at all I'm not sure which codes those buttons would have.

Colgate, 2015-01-19:
Were you able to get the volume up/down working for the k65? I am trying to determine if that's the issue, or if I'm running into the same issue as #12.

ccMSC, 2015-01-19:
Try opening a terminal and running `echo notify all > /tmp/ckb1/cmd && cat /tmp/ckb1/notify0` then press the volume buttons. You should see something like this:
`key +volup
key -volup
key +voldn
key -voldn`

If you see something other than `volup` and `voldn` then let me know what they are (or if you don't see anything printed at all). If you DO see those then it's probably the same issue.

Colgate, 2015-01-19:
They did print out, so I suppose it must be an OSX issue.

ccMSC, 2015-01-19:
Okay, well at least I know the key is detected correctly. Two other ideas:
1) Do the volume keys work when you're not using the driver?
2) Did you try downloading the old commit that I mentioned in #12? In the commit immediately after it I made some changes to how the volume keys are handled. On my mac, they weren't working before that, but they are working now. I'm wondering if it might be the opposite for some people.

Colgate, 2015-01-19:
Looks like they work fine when the daemon is stopped. Just downloaded the older commit and am about to package it for a test. I'll update when I've tested.

Colgate, 2015-01-19:
Looks like that commit was authored before you added in k65 support, and I have too little exposure to C to make a successful regression of the k65 commits. For now, I'm perfectly happy that the remaining 99% of the keyboard works as intended, and maybe I'll try another build with k65 integration when it's not 2:30AM

richlewis42, 2015-01-19:
Hi, sorry it took so long, I've been away from my desk for ages.  I checked the build and it still works.  

Regarding the volume up and down buttons, they work fine for me (increasing and decreasing the volume) both when the daemon is running and when it is terminated.  
They do not trigger the animations. 
They DO light up when the ripple and wave animations are activated (in the wrong order). 

The mute and the brightness buttons function correctly.
They trigger animations, again in the wrong order. 
They light up when ripple and wave animations are used

Implementing the proper K65 layout would fix the order - I could have a go at this, but it will probably take me a while to get around to it.

ccMSC, 2015-01-20:
@Colgate As of commit 00bf5997503ddf99ffd4cb53dd4b751dc9665385 the old key events should be back now, in addition to the new ones. Try that out and see if it works.

@richlewis42 This makes sense since the volup/voldn keys don't exist in the normal LED layouts. If you turn off animations and set the key colors individually, can you tell me which keys control the volume buttons? (I'm guessing it's one of the media keys)

Colgate, 2015-01-22:
Just noticed you addressed this. Tried on the new build, but they're still not working for me, and they do not trigger animations either.

One thing worth mentioning is that on the UI diagram, these 2 buttons are mapped to the locations of Stop and Previous

VolDown = Previous
VolUp = Stop

Not sure if that's part of the problem, or if that's simply how you're doing it to handle the shape differences in the 2 boards, but I thought I'd mention it just to be safe.

ccMSC, 2015-01-22:
@Colgate that's actually exactly what I wanted to know, thanks

Hopefully I did this right: ckb should now support the K65 layout. I added the volume keys and moved the other buttons around so they should animate properly.

I've also changed the input code for the volume keys yet again. Dunno if it will do any good or not, but the old code was pretty hacky so this should rule out any error if the event structure was somehow different. If they're still not working then I give up on them for now.

Btw, does the Fn key on the K65 work, or does it just act as a second Cmd key? I'm curious whether those functions are software or hardware based

Colgate, 2015-01-22:
So volume and mute buttons are now correctly ordered for Raindrop and Ripple (the 2 animations I use) and can trigger animations, but no such luck with them actually controlling volume.

Checking on notify, I did notice when using vol-up, it adds a ^_, and triggers an error tone when used over Finder and some other applications.

I'm still satisfied that it /mostly/ works, and it's not much of a hassle to use the onscreen controls or lean over and use the builtin volume buttons on my macbook. It's a work computer, so my biggest concern is that the main portion of the keyboard functions properly (which it does).

Fn works to trigger animations, but I'm not seeing that it does anything else, and it throws nothing for notify.

Colgate, 2015-01-22:
Update, had to restart system, restarted ckb and daemon and volume down worked for a single press.

ccMSC, 2015-01-23:
Huh. Well that's something, at least. I think the ^_ in the terminal was caused by generating both a regular keypress and a media event. I checked again and the OS definitely doesn't do that, so I set it to just create a media event. Don't know if that will solve the problem or not, but that's all I can think of for now.

Colgate, 2015-01-24:
Works!

ccMSC, 2015-01-24:
Well I'll be. Alright, sounds like this is solved for now. Thanks for your help guys :)

closed 2015-01-24 by ccMSC