ccMSC/ckb #12: Volume button not working when using app

BrendanMiller, 2015-01-12:
Hi there! I appreciate all of the help so far, but I'm still experiencing a few issues.

My biggest concern is that when I have the app open, the volume button on the keyboard doesn't work. It works fine after I shut the app down and sometimes I have to restart the computer. 

Second, should the keyboard settings stay after I shut down the app? I have to keep the app open in order to have any effects stay on my keyboard. When I save to hardware, I don't know if it's doing anything. The switches in the back to change profiles doesn't seem to work, and the keyboard starts to malfunction or just stop working when I do.

Thanks!

ccMSC, 2015-01-13:
The animations are all software based, so no, they won't work after closing the app. Hardware save only stores the base colors on the device.

I'm not sure what would cause the volume keys to malfunction, though. Could you download commit e951a3c46a7eb5e5fe8ac89818ae3601d1fc2100 and tell me if they work correctly with that version?

ccMSC, 2015-01-24:
ok, I *think* this should be fixed now. If it's not then go ahead and open a new ticket.

closed 2015-01-24 by ccMSC