ccMSC/ckb #13: K70 non rgb volunteer

th3fallen, 2015-01-23:
I'll volunteer to help out if you're still interested, im on mac if that helps.

ccMSC, 2015-01-24:
Ok, thanks. There's one or two things I need to look into first, due to USB differences between OSX and Linux, but I should have a test build ready in a couple of days. I'll update you when I do.

ccMSC, 2015-01-29:
It's up! As of commit 5aee30b0e0ce1fcd53a631d4dd2b610cfaca7965 the non-RGB K70 should be supported on OSX now. The ckb app won't really do anything for you yet, as the options all pertain to the non-RGB keyboards (I'm working on that, though - it should have proper support for them soon), but the daemon at least should work.

Let me know if you see any "Warning: got unknown handle" messages as the keyboard probably won't connect if that happens. Also let me know if there are any keys that don't produce input, especially if they're accompanied by a "Got unknown key press" message. Just paste any error messages here and tell me about anything that doesn't work.

th3fallen, 2015-01-29:
Will this by chance map the media keys correctly? Play pause and stop work but not next and previous

ccMSC, 2015-01-30:
I'm not sure what the input for those looks like, so I made a branch to test it: https://github.com/ccMSC/ckb/tree/k70test

This version of the daemon should print lines with a bunch of numbers whenever you press a key. Try all of the media keys (mute button and volume wheel too) and paste the numbers you see for each key here. You'll have to right-click and select Copy since pressing Cmd+C will spam the output with more numbers.

Btw, can you also tell me if the Program (the one above F12), Brightness, or Win Lock keys produce any output with this version? They can't be controlled via software as far as I know, but I'd just like to confirm it.

th3fallen, 2015-01-30:
Not sure if im doing something wrong but i cant compile these, when i load up the ckb.pro file i get Could not find qmake configuration file default.
Error while parsing file /Users/{username}/Downloads/ckb-master/ckb.pro. Giving up.

Ideas?

ccMSC, 2015-01-31:
Neither the master branch nor the k70test branch compiles? Try re-installing Qt and see if that makes a difference. I'd also recommend opening up Xcode at least once to see if it needs to install anything (you should see a "Welcome to Xcode" screen if it's set up properly).

th3fallen, 2015-02-02:
did that still same issue...

ccMSC, 2015-02-03:
Hmm... this is the only relevant Google result I could find: http://qt-project.org/forums/viewthread/41305
I'm guessing that a system reinstall isn't a good option, but maybe try the suggestions there (running install_name_tool, trying to build anyway) and see if that does anything. The only other thing I can think of is to reinstall Qt one more time and use the custom installation to make sure it's installing kits for Desktop development (it should have x86_64-clang or something to that effect). I've also heard that some people couldn't get Qt working unless it was installed to `/Applications/Qt`, instead of the default `/Users/<username>/Qt`. I'm out of ideas other than that. We may need to wait for Xiy to get the binary builds working.

th3fallen, 2015-02-06:
Sorry for the late reply, been being pounded at work, and yeah still cant get it working guess i'll wait for the binary builds... :frowning:

ccMSC, 2015-02-07:
Hmm alright. I haven't heard from Xiy in a while, I may try to get the binaries working on my own if he doesn't show up. I'll let you know when there's a working one.

ccMSC, 2015-03-05:
If you're still interested in this, there's a new binary version that should hopefully work for you. I *think* it supports the non-RGB K70, though I can't be sure since I haven't had anyone else to test it.

th3fallen, 2015-03-05:
Sure I'll check it out tomorrow

closed 2015-06-09 by ccMSC