ccMSC/ckb #16: Cannot build on OS X

paradox460, 2015-02-04:
When I attempt to build the project in QT5 on OS X 10.10, I cannot run the resultant executable.

Upon loading the project, QT displays the following error in General Messages:

```
Automatic type dump of QML module failed.
Errors:
"/Applications/Qt/5.4/clang_64/bin/qmlplugindump" crashed.
Arguments: --builtins
dyld: Library not loaded: /work/build/______________________________PADDING______________________________/lib/QtQuick.framework/Versions/5/QtQuick
  Referenced from: /Applications/Qt/5.4/clang_64/bin/qmlplugindump
  Reason: image not found

```

The build fails, with the following compile output:

```
13:02:33: Running steps for project ckb...
13:02:33: Configuration unchanged, skipping qmake step.
13:02:33: Starting: "/usr/bin/make" -r -w
dyld: Library not loaded: /work/build/______________________________PADDING______________________________/lib/QtCore.framework/Versions/5/QtCore
  Referenced from: /Applications/Qt/5.4/clang_64/bin/uic
  Reason: image not found
make: Entering directory `/Users/jeffsandberg/Desktop/build-ckb-Desktop_Qt_5_4_0_clang_64bit-Debug'
cd src/ckb-daemon/ && ( test -e Makefile || /Applications/Qt/5.4/clang_64/bin/qmake /Users/jeffsandberg/Desktop/ckb/src/ckb-daemon/ckb-daemon.pro -spec macx-clang CONFIG+=debug CONFIG+=x86_64 -o Makefile ) && /Applications/Xcode.app/Contents/Developer/usr/bin/make -f Makefile 
make[1]: Entering directory `/Users/jeffsandberg/Desktop/build-ckb-Desktop_Qt_5_4_0_clang_64bit-Debug/src/ckb-daemon'
/Applications/Xcode.app/Contents/Developer/usr/bin/make -f Makefile.Debug
make[2]: Entering directory `/Users/jeffsandberg/Desktop/build-ckb-Desktop_Qt_5_4_0_clang_64bit-Debug/src/ckb-daemon'
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang++ -headerpad_max_install_names -Wl,-syslibroot,/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.10.sdk -mmacosx-version-min=10.7 -Wl,-rpath,/Applications/Qt/5.4/clang_64/lib -o ../../../ckb/ckb.app/Contents/Resources/ckb-daemon debug/extra_mac.o debug/device.o debug/devnode.o debug/input_linux.o debug/input_mac.o debug/input.o debug/keyboard_us.o debug/keyboard.o debug/led.o debug/main.o debug/notify.o debug/usb_linux.o debug/usb_mac.o debug/usb.o debug/firmware.o debug/profile.o debug/keyboard_se.o debug/keyboard_gb.o debug/keyboard_de.o debug/keyboard_fr.o   -F/Applications/Qt/5.4/clang_64/lib -framework AppKit -framework CoreFoundation -framework CoreGraphics -framework IOKit -liconv -framework QtGui -framework QtCore -framework DiskArbitration -framework OpenGL -framework AGL 
make[2]: Leaving directory `/Users/jeffsandberg/Desktop/build-ckb-Desktop_Qt_5_4_0_clang_64bit-Debug/src/ckb-daemon'
make[1]: Leaving directory `/Users/jeffsandberg/Desktop/build-ckb-Desktop_Qt_5_4_0_clang_64bit-Debug/src/ckb-daemon'
cd src/ckb/ && ( test -e Makefile || /Applications/Qt/5.4/clang_64/bin/qmake /Users/jeffsandberg/Desktop/ckb/src/ckb/ckb.pro -spec macx-clang CONFIG+=debug CONFIG+=x86_64 -o Makefile ) && /Applications/Xcode.app/Contents/Developer/usr/bin/make -f Makefile 
make[1]: Entering directory `/Users/jeffsandberg/Desktop/build-ckb-Desktop_Qt_5_4_0_clang_64bit-Debug/src/ckb'
/Applications/Xcode.app/Contents/Developer/usr/bin/make -f Makefile.Debug
make[2]: Entering directory `/Users/jeffsandberg/Desktop/build-ckb-Desktop_Qt_5_4_0_clang_64bit-Debug/src/ckb'
/Applications/Qt/5.4/clang_64/bin/uic ../../../ckb/src/ckb/mainwindow.ui -o ui_mainwindow.h
make[2]: *** [ui_mainwindow.h] Trace/BPT trap: 5
make[1]: *** [debug] Error 2
make: *** [sub-src-ckb-make_first] Error 2
make[2]: Leaving directory `/Users/jeffsandberg/Desktop/build-ckb-Desktop_Qt_5_4_0_clang_64bit-Debug/src/ckb'
make[1]: Leaving directory `/Users/jeffsandberg/Desktop/build-ckb-Desktop_Qt_5_4_0_clang_64bit-Debug/src/ckb'
make: Leaving directory `/Users/jeffsandberg/Desktop/build-ckb-Desktop_Qt_5_4_0_clang_64bit-Debug'
13:02:33: The process "/usr/bin/make" exited with code 2.
Error while building/deploying project ckb (kit: Desktop Qt 5.4.0 clang 64bit)
When executing step "Make"
13:02:33: Elapsed time: 00:00.
```

ccMSC, 2015-02-05:
```
dyld: Library not loaded: /work/build/______________________________PADDING______________________________/lib/QtCore.framework/Versions/5/QtCore
  Referenced from: /Applications/Qt/5.4/clang_64/bin/uic
  Reason: image not found
```

That line seems very odd to me, like something wasn't installed properly. Make sure Xcode is up to date and opens correctly on its own. After that I'd try reinstalling Qt and rebooting the computer.

paradox460, 2015-02-05:
I finally got it built, after re-installing Qt with all (including android libs) on my machine

closed 2015-02-05 by paradox460