ccMSC/ckb #17: Error installing OSX Binaries

jaufgang, 2015-02-05:
Following instructions for installing OSX binaries, and I get this:

>$sudo ckb.app/Contents/Resources/ckb-daemon 
>dyld: Library not loaded: /Applications/Qt/5.4/clang_64/lib/QtGui.framework/Versions/5/QtGui
>  Referenced from: /Applications/ckb.app/Contents/Resources/ckb-daemon
>  Reason: image not found

I dug around and found that this library is included with the ckb distribution at the path
`/Applications/ckb.app/Contents/Frameworks/QtGui.framework/Versions/5\QtGui`, and I think it should be trying to load it from there, instead of the `Applications/Qt folder`, for users (like me) who don't already have the Qt libraries installed separately.

ccMSC, 2015-02-06:
Yeah, it's a problem with the way Xiy built the binary package. I haven't heard from him in a while so I'm not sure if he figured it out or not. Meanwhile, I think you can fix it by installing Qt to /Applications/Qt, with or without Xcode installed. Let me know if that works.

xiy, 2015-02-12:
Hi guys, apologies for going AWOL - a bit of family trouble. I've messed around with `macdeployqt` to try and get the frameworks recognised in the app bundle but no joy. Gonna give it another go today.

ccMSC, 2015-02-12:
Still giving me the same error unfortunately:
```
dyld: Library not loaded: /Applications/Qt/5.4/clang_64/lib/QtGui.framework/Versions/5/QtGui
  Referenced from: /Applications/ckb.app/Contents/Resources/ckb-daemon
  Reason: image not found
Trace/BPT trap: 5
```
I just realized though...neither the daemon nor the animations have any dependencies on Qt (just C libraries and some Apple frameworks). You should be able to remove the Qt frameworks from the link process without any problems. If there's something I need to do to the .pro files to accomplish that, let me know.

xiy, 2015-02-13:
I expect that might actually be the issue - can you try opening just ckb.app?

ccMSC, 2015-02-13:
ckb,app works fine. (other than missing animations, since those don't run either)

ccMSC, 2015-02-17:
I think I figured this out: The solution is to use `QT -= core` and `QT -= gui` in the project files. Removed them as of commit 566d3cb71d3986ec423c71261dfd7e680fb370e0. Try compiling it now and see if the dependencies are gone.

ccMSC, 2015-03-05:
Works correctly on my test machine with the latest binary (alpha 0.0.42, link in readme). If it's still not running for any reason, please open a new issue.

closed 2015-03-05 by ccMSC