ccMSC/ckb #18: Conflicts with Karabiner

paradox460, 2015-02-05:
My previous issue was resolved, but upon loading the daemon, I noticed a new one.

I use the program Karabiner on OS X to move some keys around, i.e make the windows key function as the mac's Alt/Option key, the menu key to be FN, etc.

Unfortunately, as soon as the daemon loads, it seems to interfere with the normal functionality of Karabiner.

Not sure if this is easily fixable, but hopefully something can be done

----

PS, I actually think that the software has potential to be better than Corsair's official software. The gradient editor was far easier to use (partially due to the fact that I can specify absolute positions for stops, partially because it comes with some sane defaults)

ccMSC, 2015-02-06:
It looks like Karabiner works by remapping the keys before they reach the operating system. Unfortunately, since ckb does the same thing, I don't think there's any way to make them compatible. ckb is going to have key rebinding soon, though, so hopefully that should provide the customization that you want.

closed 2015-02-06 by ccMSC

paradox460, 2015-02-06:
That should do it for me. Thanks