ccMSC/ckb #19: Retina icon in menubar

grosen, 2015-02-08:
OS X

ccMSC, 2015-02-09:
Is your version of Qt up to date? According to this it was fixed in version 5.4: https://bugreports.qt.io/browse/QTBUG-33441

I don't have a retina display so I can't test it, but the image file is more than large enough.

ccMSC, 2015-02-22:
Hello? Can you tell me what version of Qt you're using?

You can find it in Qt Creator's "About" window. For instance, mine says: `Based on Qt 5.4.0 (GCC 4.9.2, 64 bit)`

jashephe, 2015-03-21:
Not the OP, but the bundled QT version does appear to be 5.4.0.  I've attached a few screenshots of what it looks like.  It's not just the menubar; the whole application is running in Low-DPI mode.

[image]
The menubar icon compared to other Hi-DPI icons.

[image]
The application window compared to a Hi-DPI chrome window.

I attempted a hack-y fix in the style of [this suggestion](http://apple.stackexchange.com/a/69052/53142) but wasn't able to get the Info.plist flag to do anything.  I'm not too familiar with Qt; perhaps a Hi-DPI flag of some sort needs to be set there?

ccMSC, 2015-03-23:
Ah, I thought that might be the case.

I think it needs to be enabled within the app itself - see if it's working now (v0.0.49)

jashephe, 2015-03-23:
Sorry for the trouble, but are pre-built versions of v0.0.49 available anywhere?

ccMSC, 2015-03-24:
Not right now, sorry. @xiy compiles them, not me

xiy, 2015-03-24:
I'll build v0.0.49 tomorrow (GMT).

ccMSC, 2015-03-24:
Actually I just realized I forgot to add stuff to `info.plist` - bumped the version to 0.0.50, compile that instead. Thanks :)

xiy, 2015-03-27:
@ccMSC, @jashephe 

New release :)

https://github.com/ccMSC/ckb/releases/tag/0.0.52

jashephe, 2015-03-27:
And indeed, that seems to do it:

[image]

[image]

I'm not the OP, so I can't close this, but it sure looks resolved to me. Cheers!

closed 2015-03-27 by grosen