ccMSC/ckb #20: K95 RGB won't initialize due to hwloadprofile() failure

lathiat, 2015-02-09:
I have a K95 RGB on Linux (Ubuntu 15.04, 3.18.0-12-generic).

When launching ckb-daemon, it gets stuck here and never succeeds:

lathiat@ubuntu:~/src/ckb$ sudo ./bin/ckb-daemon
[sudo] password for lathiat:
ckb Corsair Keyboard RGB driver alpha-v0.0.35
Setting FPS to 30
Setting default layout: us
Root controller ready at /dev/input/ckb0
Connecting Corsair K95 RGB Gaming Keyboard (S/N: 06008015AE3618E3530D4C22F5001946)
Error: usbinput (profile.c:307): Operation not permitted
USB failure. Attempting reset (1 of 25)...
Error: usbinput (profile.c:307): Operation not permitted
Attempting reset (2 of 25)...

(this goes on forever)

If I stub out hwloadprofile() to just return 0, the keyboard works and the RGB stuff works also.
I've never used this on Windows, so i'm not 100% sure what these profiles are doing but if it's some kind of memory perhaps it's not initialised? Not sure.

lathiat, 2015-02-09:
(gdb) run
Starting program: /home/lathiat/src/ckb/bin/ckb-daemon
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
ckb Corsair Keyboard RGB driver alpha-v0.0.35
Setting FPS to 30
Setting default layout: us
Root controller ready at /dev/input/ckb0
[New Thread 0x7ffff731c700 (LWP 10188)]
[New Thread 0x7ffff6b1b700 (LWP 10189)]
Connecting Corsair K95 RGB Gaming Keyboard (S/N: 06008015AE3618E3530D4C22F5001946)
[New Thread 0x7ffff631a700 (LWP 10190)]
[Switching to Thread 0x7ffff731c700 (LWP 10188)]

Breakpoint 1, hwloadprofile (kb=0x616950 <keyboard+1168>, apply=1) at profile.c:307
307             if(!usbinput(kb, in_pkt)){
(gdb) bt
#0  hwloadprofile (kb=0x616950 <keyboard+1168>, apply=1) at profile.c:307
#1  0x000000000040d458 in setupusb (kb=0x616950 <keyboard+1168>, vendor=6940, product=6929) at usb.c:129
#2  0x000000000040c3fe in openusb (dev=0x7ffff0000e70, vendor=6940, product=6929) at usb_linux.c:267
#3  0x000000000040c753 in udevenum () at usb_linux.c:357
#4  0x000000000040c7dc in udevmain (context=0x0) at usb_linux.c:371
#5  0x00007ffff79b30a5 in start_thread (arg=0x7ffff731c700) at pthread_create.c:309
#6  0x00007ffff76e090d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111
(gdb) print kb
$1 = (usbdevice *) 0x616950 <keyboard+1168>
(gdb) print in_pkt
$2 = "\000\000\000\000\000\000\000\000\000/l\240.\027ǩ\340\274\061\367\377\177\000\000;\346@\000\000\000\000\000(\000\000\000@\000\000\000\000\061a\000\000\000\000\000hja\000\000\000\000\000\210ja\000\002\000\000"
(gdb) next
Error: usbinput (profile.c:307): Operation not permitted
308                 free(hw);
(gdb) break usb_linux.c:37
Breakpoint 2 at 0x40b7d1: file usb_linux.c, line 37.
(gdb) cont
Continuing.
USB failure. Attempting reset (1 of 25)...

Breakpoint 2, _usbinput (kb=0x616950 <keyboard+1168>, message=0x7ffff731bc50 "", file=0x411496 "profile.c", line=307) at usb_linux.c:37
37          if(res <= 0){
(gdb) print res
$4 = -1
(gdb) up
#1  0x000000000040f8c4 in hwloadprofile (kb=0x616950 <keyboard+1168>, apply=1) at profile.c:307
307             if(!usbinput(kb, in_pkt)){
(gdb) bt
#0  _usbinput (kb=0x616950 <keyboard+1168>, message=0x7ffff731bc50 "", file=0x411496 "profile.c", line=307) at usb_linux.c:37
#1  0x000000000040f8c4 in hwloadprofile (kb=0x616950 <keyboard+1168>, apply=1) at profile.c:307
#2  0x000000000040d742 in _resetusb (kb=0x616950 <keyboard+1168>, file=0x411219 "usb.c", line=201) at usb.c:189
#3  0x000000000040d7e5 in usb_tryreset (kb=0x616950 <keyboard+1168>) at usb.c:201
#4  0x000000000040c43b in openusb (dev=0x7ffff0000e70, vendor=6940, product=6929) at usb_linux.c:273
#5  0x000000000040c753 in udevenum () at usb_linux.c:357
#6  0x000000000040c7dc in udevmain (context=0x0) at usb_linux.c:371
#7  0x00007ffff79b30a5 in start_thread (arg=0x7ffff731c700) at pthread_create.c:309
#8  0x00007ffff76e090d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111
(gdb) down
#0  _usbinput (kb=0x616950 <keyboard+1168>, message=0x7ffff731bc50 "", file=0x411496 "profile.c", line=307) at usb_linux.c:37
37          if(res <= 0){
(gdb) print res
$5 = -1
(gdb) l
32      int _usbinput(usbdevice* kb, uchar* message, const char* file, int line){
33          if(!IS_ACTIVE(kb))
34              return -1;
35          struct usbdevfs_ctrltransfer transfer = { 0xa1, 0x01, 0x0300, 0x03, MSG_SIZE, 50, message };
36          int res = ioctl(kb->handle, USBDEVFS_CONTROL, &transfer);
37          if(res <= 0){
38              printf("Error: usbinput (%s:%d): %s\n", file, line, res ? strerror(-res) : "No data read");
39              return 0;
40          }
41          if(res != MSG_SIZE)

lathiat, 2015-02-09:
I'm going to guess "save to hardware" might make this go away.  But I'm going to avoid that until you've had a chance to debug with me so we can maybe fix the first run issue :)

ccMSC, 2015-02-09:
"Operation not permitted" is usually a USB communication error. Did you try the suggestion about the kernel cmdline parameters?

I don't think that running a hwsave would fix the issue. There's a default profile built in to the firmware that it should be loading. What firmware version is your keyboard running? (If ckb's firmware detection isn't working, you can check it with `lsusb -vd 1b1c: | grep bcdDevice`)

closed 2015-06-09 by ccMSC

lathiat, 2015-06-09:
Upgrading firmware to 1.20 fixed this, it was still a problem right up until I did that and would never work.  So likely an older firmware issue.

lathiat, 2015-06-09:
I think I had one of the very first devices (direct from Corsair), and I had never used the Windows tool until the other day.

ccMSC, 2015-06-09:
Aaah okay. I closed the issue due to inactivity but that's good to hear.