ccMSC/ckb #24: Will this work on wayland?

Rabcor, 2015-02-18:
I'm wondering about 4 things, do all the G-Keys work yet?, are the generic keyboard features/buttons all working? Do the volume wheel and media keys work? And will this driver work on wayland?

btreeinfinity, 2015-02-18:
So far everything appears to work on Fedora21/Gnome-shell/Wayland. Should work on other distros. Can you go ahead and install and test? I can help with the install if needed.

Rabcor, 2015-02-19:
I would but I don't have the keyboard yet :p, that's why I'm asking instead of trying.

ccMSC, 2015-02-19:
Should work just fine. You might need to install Qt5 Wayland packages but that's it.

All of the keyboard's (hardware) functions are currently supported. Options for key rebinding are a bit limited at the moment but I'm still working on them. But yes, you can use the G-keys.

Rabcor, 2015-02-19:
Awesome, thanks :)

closed 2015-06-09 by ccMSC

Rabcor, 2016-01-21:
I just tried it on wayland, I can confirm that ckb works on wayland, but only through xwayland. If xwayland is disabled it ckb (and even ckb --background) gives a "could not connect to display" error.