ccMSC/ckb #26: error activating daemon

rsoul, 2015-02-22:
The app always warns me that the daemon is inactive. I have followed instructions and pasted the command sudo /Applications/ckb.app/Contents/Resources/ckb-daemon but terminal replies that:

dyld: Library not loaded: /Users/rsoul/Qt5.4.0/5.4/clang_64/lib/QtCore.framework/Versions/5/QtCore
  Referenced from: /Applications/Qt/5.4/clang_64/lib/QtGui.framework/Versions/5/QtGui
  Reason: image not found

i tried updating xcode and qt but that doesn't look like it's the issue. Any ideas?

By the way great job with the project, i really appreciate your work! You totally deserved a donation!

ccMSC, 2015-02-23:
Hmm...it really shouldn't depend on those libraries at all. I added some more lines to the project files that might fix it. I'd recommend completely deleting the ckb directory (also delete any "build-ckb" directories created by Qt) and trying again with the latest commit.

btreeinfinity, 2015-03-03:
Troubleshooting (Linux)

If you have problems connecting the keyboard to your system (keyboard doesn't respond, ckb-daemon doesn't recognize it or can't connect the keyboard), try adding the following to your kernel's cmdline:

K65 RGB: usbhid.quirks=0x1B1C:0x1B17:0x20000000
K70: usbhid.quirks=0x1B1C:0x1B09:0x20000000
K70 RGB: usbhid.quirks=0x1B1C:0x1B13:0x20000000
K95 RGB: usbhid.quirks=0x1B1C:0x1B11:0x20000000
For instructions on adding cmdline parameters in Ubuntu, see https://wiki.ubuntu.com/Kernel/KernelBootParameters

ccMSC, 2015-03-05:
If you still haven't had any luck compiling it, the latest binary version should work correctly.

closed 2015-06-09 by ccMSC