ccMSC/ckb #29: K95 Non-RGB Volunteer

Persata, 2015-03-02:
Hi!

I've got a non-RGB K95 I'm using on OSX, willing to offer what help I can to get it functional - I really miss the macro keys for coding.

Here's the system summary information for the K95:

[image]

Tried adding variables (e.g. `P_K95_NRGB` similar to `P_K70_NRGB`), updated `IS_RGB()` etc and recompiled in Qt Creator, but no dice - running the daemon and the app just stopped keyboard input altogether. Not sure what else to try, I'm more PHP & Javascript than C++ I'm afraid.

Let me know how I can be of assistance!

ccMSC, 2015-03-03:
Do you see any output from the daemon? (I assume it prints "Connecting Corsair K95"...does it get beyond that or does it just stay there?) I'll start a new branch tomorrow to print out some more info about the keyboard and (hopefully) figure out what I need to do to support it

Persata, 2015-03-03:
This is the only output I get from the daemon:

```
ckb Corsair Keyboard RGB driver alpha-v0.0.41
Setting FPS to 30
Setting default layout: gb
Root controller ready at /tmp/ckb0
```

The keyboard input doesn't work until I terminate the daemon process. Not sure what else to try, so I'll await your branch and give that a whirl =)

ccMSC, 2015-03-04:
k95test branch is up: https://github.com/ccMSC/ckb/tree/k95test

Looks like I forgot to handle serial/firmware numbers for the non-RGB boards, which explains the lack of messages (I haven't been able to test the non-RGB boards on OSX so far because the only person who volunteered wasn't able to compile it). I think it should work now, but I included some extra debug code in case it doesn't. So if the key input still isn't working, paste the output from the new daemon here.

Persata, 2015-03-04:
Ok, I have compiled the k95test branch in Qt and started the daemon and this is the only output:

```
ckb Corsair Keyboard RGB driver alpha-v0.0.42
Setting FPS to 30
Setting default layout: gb
Root controller ready at /tmp/ckb0
```

No extra logging or debugging info. If I run ckb.app, it also says "No devices connected".

However, I can get keyboard input when the daemon is running now which is a definite improvement. Anything else I should try?

Thanks for taking the time to work on this!

ccMSC, 2015-03-05:
Sure you got the right branch? It should say `alpha-v0.0.42+k95test_1` if compiled from scratch. The version in the master branch won't connect the keyboard yet.

Persata, 2015-03-05:
You're right indeed - seems I hadn't checked the branch out properly, re-did it, compiled, ran, here's the output:

```
ckb Corsair Keyboard RGB driver alpha-v0.0.42+k95test_1
Setting FPS to 30
Setting default layout: gb
Root controller ready at /tmp/ckb0
Connecting Corsair K95W Gaming Keyboard (S/N: 1b1c:1b0804-NoID)
Got handle I: 8, O: 1, F: 0
Got handle I: 15, O: 0, F: 0
Got handle I: 4, O: 0, F: 0
Warning: Got unknown handle (I: 4, O: 0, F: 0)
```

Keyboard input stops working after the daemon is started like it did before until I kill the daemon process.

Anything else to test?

ccMSC, 2015-03-06:
Just pushed a new version that should recognize the handle. See if it works now.

Persata, 2015-03-06:
Here's what I'm seeing now:

```
ckb Corsair Keyboard RGB driver alpha-v0.0.42+k95test_1
Setting FPS to 30
Setting default layout: gb
Root controller ready at /tmp/ckb0
Connecting Corsair K95W Gaming Keyboard (S/N: 1b1c:1b0804-NoID)
Got handle I: 4, O: 0, F: 0
Got handle I: 8, O: 1, F: 0
Got handle I: 15, O: 0, F: 0
Device ready at /tmp/ckb1
```

and then a whole lot of this repeated:

```
Report ID 0: 000000000000000
```

When I press a key, I get the following (different for each key)

```
Report ID 0: 001000000000000
Report ID 0: 0002000000000000
Report ID 0: 0008000000000000
```

When I start the ckb application itself, the daemon reports:

```
Error: usbdequeue (main.c:220): Got return value 0xe000404f
Attempting reset...
```

and then repeatedly this:

```
Error: usbdequeue (firmware.c:23): Got return value 0xe000404f
```

The app itself looks like this:

[image]

But I am unable to click any of the buttons, and after a short while it force closes.

With the daemon running, my Mac no longer processes any keyboard input until I kill the daemon process.

Is this helpful?

ccMSC, 2015-03-07:
Whoops, looks like there was still some code trying to send RGB messages to non-RGB boards. k95test_3 is up, it should fix that.

So it seems like the daemon is receiving input reports but doesn't know how to process them. Can you send me the output from pressing the following keys: A, S, D, Left Ctrl, Left Shift? You should see two messages for each key, one when pressed and one when released. The second one should just be a bunch of zeroes.

Persata, 2015-03-09:
Sorry for the delay, I don't have access to this mac over the weekend.

Here's the output for each of those keys (removed all the blank / all zero output):

```
A: 00 10 00 00 00 00 00 00 00 00 00 00 00 00 00
```

```
S: 00 00 00 40 00 00 00 00 00 00 00 00 00 00 00
```

```
D: 00 80 00 00 00 00 00 00 00 00 00 00 00 00 00
```

```
Left Ctrl: 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```

```
Left Shift: 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```

ccMSC, 2015-03-11:
k95test_4 is up - hopefully it should respond to key input this time!

Can you post the daemon output for each of the 18 G keys as well? I don't think they'll be recognized yet. All the other keys should work.