ccMSC/ckb #37: K95 RGB Volenteer

Tazmain, 2015-03-11:
Hi, I want to volenteer to help test on linux, I am running arch with the latest version of plasma5, any other DE is also no problem for me. Once I have a bit more C and qt under my belt I will volenteer to help with bug fixing.

ccMSC, 2015-03-12:
The RGB K95 should be fully supported already - just compile it and run. Let me know if you have any problems :)

Tazmain, 2015-03-12:
I only have a little trouble when booting from Windows to Linux with the profiles stored on the Keyboard. The default lightning profiles you have, are they animated? Mine are just static.

ccMSC, 2015-03-12:
You can't store animations on the keyboard, unfortunately. You'll have to recreate them.

Tazmain, 2015-03-12:
Do the light animation work in linux? Right now I can only set colours. Also the Ckb demo profiles the rainbow one is only one colour, is that correct? 
How did you manage to fix the caps lock status indicator issue?

Tazmain, 2015-03-12:
So I ran into some little issues, one if you open ckb in a terminal and then shutdown and when asked to close the window you say yes, the system has trouble stopping the daemon. Next when you open ckb in a terminal and quit it and then try to open it again you get this error: "QCoreApplication::postEvent: Unexpected null receiver" Trying to launch it a few times will get past this or finding the process and killing it.

Tazmain, 2015-03-12:
Seems the animation folder also doesn't get linked properly as the ckb GUI interface keeps complaining that it can't find it.

Tazmain, 2015-03-12:
Also it seems that I cannot change the colour of the \| key it seems to have a all 3 LED's on, casuing a white colour. When all a profile with all the LED's turned off is selected it does respond to that.

ccMSC, 2015-03-13:
\| being stuck white means you need to change the keyboard layout - see Device > Layout.

It sounds like something didn't build right when you compiled ckb. What did you use to compile it and can you post the output?

Tazmain, 2015-03-14:
I compiled it again, I am adding the output and warnings. 
Warinings : http://pastie.org/10025060 
Buildoutput: http://pastebin.com/Agg6Jy9e
Chaning Device Layout fixed that issue, Just a curious question with the rainbow is the aswd and arrow keys suppose to stay white?

Tazmain, 2015-03-14:
Can I get your permission to make a AUR package for this? Just a thing to doa git clone and build for arch users?

ccMSC, 2015-03-15:
Huh. Well it sounds like it worked fine this time, no idea why it didn't like properly the first time around. And yeah, the WASD/arrow keys are supposed to be white on the demo profile.

I think someone already submitted an AUR package: https://aur.archlinux.org/packages/ckb-git/
But you could ask them about it if you want to maintain it, it doesn't look like anything's been done since it was first uploaded. Anyway, thanks for the offer :)

Tazmain, 2015-03-15:
Well the lightning effects don't link properly to the ckb that gets copied to /bin only if I start ckb from the location it was compiled does it work. I found that if I open ckb and close it or exit and then shutdown the daemon has trouble stopping, however if I boot and don't open ckb and shutdown then there is no issues with the daemon.

ccMSC, 2015-03-16:
That actually makes sense since it scans the current working directory, not necessarily the path that it was installed to.

I have noticed the OS doesn't always reconnect the keyboard when ckb-daemon exits, but I don't think I can do anything about it (moving the poll rate switch or unplugging/replugigng the keyboard usually fixes it, though). There's no real reason not to run the daemon as a system service anyway.

Tazmain, 2015-04-03:
The daemon it taking a while to pick up the keyboard. I have also noticed when rebooting between windows and linux sometimes both the windows and linux client fail to see the keyboard thus they cannot control the lights at all. I have to unplug and replug the keyboard.

ccMSC, 2015-04-04:
Sounds like a firmware/BIOS issue. Corsair's promised a new firmware in the next 4-8 weeks, fingers crossed...

closed 2015-06-09 by ccMSC