ccMSC/ckb #39: ckb-daemon: No route to host

micaelbergeron, 2015-03-13:
Hi, great work so far!
I have an intermittent problem with the ckb-daemon, and the K70 RGB in general.

It seems to be unable to claim the device most of the time (like 3/4 reboots).
I'm also unable to use the keyboard in the GRUB (did that happen to anyone?)

Hope this help you find out the problem.

**lsusb**
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 005: ID 1038:1369 SteelSeries ApS 
Bus 001 Device 004: ID 046d:c22d Logitech, Inc. G510 Gaming Keyboard
Bus 001 Device 003: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 1b1c:1b13 Corsair 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

**dmesg**
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-316-x86_64 root=UUID=258d1262-5ac7-4355-aab2-e89303f8de0e rw quiet splash usbhid.quirks=0x1B1C:0x1B13:0x20000000
[    0.000000] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-316-x86_64 root=UUID=258d1262-5ac7-4355-aab2-e89303f8de0e rw quiet splash usbhid.quirks=0x1B1C:0x1B13:0x20000000
[    0.415581] usbcore: registered new interface driver usbfs
[    0.415593] usbcore: registered new interface driver hub
[    0.415826] usbcore: registered new device driver usb
[    0.736234] usb 1-1: new high-speed USB device number 2 using ehci-pci
[    0.966175] usb 2-1: new high-speed USB device number 2 using ehci-pci
[    1.249456] usb 3-3: new full-speed USB device number 2 using xhci_hcd
[   11.470868] usb 3-3: string descriptor 0 read error: -110
[   16.468211] usb 3-3: can't set config #1, error -110
[   16.714746] usb 1-1.1: new high-speed USB device number 3 using ehci-pci
[   17.077823] usb 1-1.1.3: new full-speed USB device number 4 using ehci-pci
[   17.257848] usb 1-1.1.4: new full-speed USB device number 5 using ehci-pci
[   17.386821] usbcore: registered new interface driver usbhid
[   17.386824] usbhid: USB HID core driver
[   17.388738] input: Logitech G510 Gaming Keyboard as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.1/1-1.1.3/1-1.1.3:1.0/0003:046D:C22D.0001/input/input17
[   17.389048] hid-generic 0003:046D:C22D.0001: input,hidraw0: USB HID v1.11 Keyboard [Logitech G510 Gaming Keyboard] on usb-0000:00:1a.0-1.1.3/input0
[   17.393318] input: Logitech G510 Gaming Keyboard as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.1/1-1.1.3/1-1.1.3:1.1/0003:046D:C22D.0002/input/input18
[   17.393733] hid-generic 0003:046D:C22D.0002: input,hiddev0,hidraw1: USB HID v1.11 Device [Logitech G510 Gaming Keyboard] on usb-0000:00:1a.0-1.1.3/input1
[   17.396366] hid-generic 0003:1038:1369.0003: hiddev0,hidraw2: USB HID v1.11 Device [SteelSeries Sensei Raw Gaming Mouse] on usb-0000:00:1a.0-1.1.4/input0
[   17.396499] input: SteelSeries Sensei Raw Gaming Mouse as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.1/1-1.1.4/1-1.1.4:1.1/0003:1038:1369.0004/input/input19
[   17.396747] hid-generic 0003:1038:1369.0004: input,hidraw3: USB HID v1.11 Mouse [SteelSeries Sensei Raw Gaming Mouse] on usb-0000:00:1a.0-1.1.4/input1
[   17.396887] input: SteelSeries Sensei Raw Gaming Mouse as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.1/1-1.1.4/1-1.1.4:1.2/0003:1038:1369.0005/input/input20
[   17.397078] hid-generic 0003:1038:1369.0005: input,hidraw4: USB HID v1.11 Keyboard [SteelSeries Sensei Raw Gaming Mouse] on usb-0000:00:1a.0-1.1.4/input2
[   17.397418] input: SteelSeries Sensei Raw Gaming Mouse as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.1/1-1.1.4/1-1.1.4:1.3/0003:1038:1369.0006/input/input21
[   17.397511] hid-generic 0003:1038:1369.0006: input,hidraw5: USB HID v1.11 Device [SteelSeries Sensei Raw Gaming Mouse] on usb-0000:00:1a.0-1.1.4/input3
[  101.397059] usb 3-3: reset full-speed USB device number 2 using xhci_hcd

**ckb-daemon**
ckb Corsair Keyboard RGB driver alpha-v0.0.45
Setting FPS to 30
Setting default layout: gb
Root controller ready at /dev/input/ckb0
Connecting  (S/N: 1b1c:1b1304-NoID)
Error: Failed to claim interface: Aucun chemin d'accès pour atteindre l'hôte cible

micaelbergeron, 2015-03-13:
Here are my notes from my tests.
 * USB 3.0 ports should be avoided.
 * Only plug the cable with the keyboard etching on it
I cannot seem to get ckb-daemon to claim the device at startup, as the device isn't listed in dmesg until I plug it back in. Only then I can get the daemon to work.

For the BIOS part, I still can't get the keyboard working unless I manually set the back switch to BIOS.
The sad part of this is that the lighting level is cycling from every keystroke, which makes this solution 1) painful 2) hard on the eyes and 3) impossible to use with ckb-daemon.

Here are outputs of dmesg, with the switch at different position, replugging the device each time.

**switch on BIOS**
[    2.350157] usb 2-1.6: usbfs: USBDEVFS_CONTROL failed cmd ckb-daemon rqt 161 rq 1 len 64 ret -110
[    2.533462] Switched to clocksource tsc
[    2.558210] NVRM: Your system is not currently configured to drive a VGA console
[    2.558213] NVRM: on the primary VGA device. The NVIDIA Linux graphics driver
[    2.558214] NVRM: requires the use of a text-mode VGA console. Use of other console
[    2.558216] NVRM: drivers including, but not limited to, vesafb, may result in
[    2.558217] NVRM: corruption and stability problems, and is not supported.
[    2.623279] usb 2-1.6: reset full-speed USB device number 3 using ehci-pci
[    3.403508] usb 2-1.6: usbfs: USBDEVFS_CONTROL failed cmd ckb-daemon rqt 161 rq 1 len 64 ret -110
[    3.676661] usb 2-1.6: reset full-speed USB device number 3 using ehci-pci
[    3.819864] usb 2-1.6: device firmware changed
[    3.820109] usb 2-1.6: USB disconnect, device number 3
[    3.886605] usb 2-1.6: new full-speed USB device number 4 using ehci-pci
[    4.034599] usb 2-1.6: unable to read config index 0 descriptor/start: -32
[    4.034602] usb 2-1.6: chopping to 0 config(s)
[    4.044269] usb 2-1.6: no configuration chosen from 0 choices

**switch at 1**
[  672.301023] usb 2-1.6: USB disconnect, device number 4
[  673.790974] usb 2-1.6: new full-speed USB device number 5 using ehci-pci
[  674.617103] input: Corsair Corsair K70 RGB Gaming Keyboard as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.6/2-1.6:1.0/0003:1B1C:1B13.0007/input/input23
[  674.617267] hid-generic 0003:1B1C:1B13.0007: input,hidraw6: USB HID v1.11 Keyboard [Corsair Corsair K70 RGB Gaming Keyboard] on usb-0000:00:1d.0-1.6/input0

**BIOS**
[  692.329479] usbhid 2-1.6:1.1: can't add hid device: -32
[  692.329813] usbhid: probe of 2-1.6:1.1 failed with error -32
[  692.333867] usbhid 2-1.6:1.2: can't add hid device: -32
[  692.334172] usbhid: probe of 2-1.6:1.2 failed with error -32
[  692.338112] usbhid 2-1.6:1.3: can't add hid device: -32
[  692.338419] usbhid: probe of 2-1.6:1.3 failed with error -32
[  692.424058] input: ckb1 as /devices/virtual/input/input24
[  692.500233] usb 2-1.6: USB disconnect, device number 5
[  693.646955] usb 2-1.6: new full-speed USB device number 6 using ehci-pci
[  694.473218] input: Corsair Corsair K70 RGB Gaming Keyboard as /devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.6/2-1.6:1.0/0003:1B1C:1B13.0008/input/input25
[  694.473386] hid-generic 0003:1B1C:1B13.0008: input,hidraw6: USB HID v1.11 Keyboard [Corsair Corsair K70 RGB Gaming Keyboard] on usb-0000:00:1d.0-1.6/input0

Keep up the good work, tell me if you need anything to test this out.

ccMSC, 2015-03-13:
I think your firmware is outdated - the latest version, v1.15, shouldn't have the key brightness issue. There are instructions for updating it in DAEMON.md, so I recommend trying that first. I also recommend updating your computer's BIOS if possible. It sounds like something is happening during your BIOS boot that prevents the keyboard from talking to the computer.

micaelbergeron, 2015-03-13:
Will do. I'll kke you updated.

fliiiix, 2015-03-16:
I have a similary problem, but with the keyboard at my PC it doesn't even boot. 
[image]

I have upgraded the firmware to the latest version and my os is Fedora 21 (the lxde spin).

closed 2015-06-09 by ccMSC