ccMSC/ckb #40: can't add hid device: -110

fliiiix, 2015-03-16:
A bit more information to https://github.com/ccMSC/ckb/issues/39#issuecomment-81568589.

If I connect the keyboard after boot it works, but with errors.

dmesg 
```
[   74.614529] usb 1-8: new full-speed USB device number 5 using xhci_hcd
[   75.300067] usb 1-8: New USB device found, idVendor=1b1c, idProduct=1b13
[   75.300077] usb 1-8: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   75.300082] usb 1-8: Product: Corsair K70 RGB Gaming Keyboard
[   75.300086] usb 1-8: Manufacturer: Corsair
[   75.300090] usb 1-8: SerialNumber: 1801501EAE3B940353758109F5001943
[   75.509350] input: Corsair Corsair K70 RGB Gaming Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:1B1C:1B13.0004/input/input19
[   75.510078] hid-generic 0003:1B1C:1B13.0004: input,hidraw3: USB HID v1.11 Keyboard [Corsair Corsair K70 RGB Gaming Keyboard] on usb-0000:00:14.0-8/input0
[  110.476161] usbhid 1-8:1.1: can't add hid device: -110
[  110.476191] usbhid: probe of 1-8:1.1 failed with error -110
[  145.441997] usbhid 1-8:1.2: can't add hid device: -110
[  145.442007] usbhid: probe of 1-8:1.2 failed with error -110
[  180.407330] usbhid 1-8:1.3: can't add hid device: -110
[  180.407360] usbhid: probe of 1-8:1.3 failed with error -110
[  180.444727] input: ckb1 as /devices/virtual/input/input20
[  180.943751] usb 1-8: usbfs: USBDEVFS_CONTROL failed cmd ckb-daemon rqt 33 rq 9 len 1 ret -110
[  181.443198] usb 1-8: usbfs: USBDEVFS_CONTROL failed cmd ckb-daemon rqt 33 rq 9 len 64 ret -110
[  182.045630] usb 1-8: usbfs: USBDEVFS_CONTROL failed cmd ckb-daemon rqt 33 rq 9 len 64 ret -110
[  182.398384] usb 1-8: reset full-speed USB device number 5 using xhci_hcd
[  182.815752] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8801febbac00
[  182.815755] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8801febbad20
[  182.815756] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8801febbad80
[  182.815757] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8800da303b88
[  182.815758] xhci_hcd 0000:00:14.0: xHCI xhci_drop_endpoint called with disabled ep ffff8800da303b40
```

Btw. Thanks for your awesome work it's a shame that corsair doesn't sponsor your work or something like that.

ccMSC, 2015-03-16:
Which of the usbhid.quirks lines are you using (if any)? Have you tried updating your BIOS or putting the keyboard into BIOS mode at boot?

fliiiix, 2015-03-17:
I tried this one 
``` 
usbhid.quirks=0x1B1C:0x1B13:0x400
```

> Have you tried updating your BIOS

As far as I know I have the latest version, but I will check it. 

> putting the keyboard into BIOS mode at boot

Not sure what this means but I will check my BIOS.

fliiiix, 2015-03-17:
I tired a few things. 

First of all there was a BIOS update, which I did now.
And the keyboard works fine for the BIOS and Grub if I coldboot it. (Which is good enough for me)

And the first time after the BIOS update everything worked fine, except much garbage output on my screen  
[image]

After that I thought nice it works like it should but nope, after a restart
[image]

And with this I can't boot my Fedora.
:

ccMSC, 2015-03-18:
So it's working fine in GRUB even after a reboot but fails as soon as Linux starts? What does ckb-daemon do when you run it after this?

There's a switch on the back of the keyboard that controlls the polling interval. If you slide it all the way to the side labelled "BIOS" it puts the keyboard into BIOS compatibility mode, which sometimes solves problems like this.

fliiiix, 2015-03-18:
> So it's working fine in GRUB even after a reboot but fails as soon as Linux starts?

Exactly, then I got the usbhid probe of 1-X:1.2 failed with error -110 which stops the entire booting of linux. 

> What does ckb-daemon do when you run it after this?

Most time it works fine, find the keyboard and start. I played around with the `usbhid.quirks` and if I use 0x4 everything looks fine but I can't use this with full disk encryption. 

> "BIOS" it puts the keyboard into BIOS compatibility mode, which sometimes solves problems like this.

I played around with this doesn't solve the issue for me.

ccMSC, 2015-03-20:
Hmm. It might be possible to get ckb-daemon into your initramfs and have it run automatically (which would solve the FDE problem). Not sure how that works on Fedora. There's not much I can do to help since the problem is with the keyboard at boot and not with my driver.

fliiiix, 2015-03-20:
Sounds right. I will close the issue for now.

closed 2015-03-20 by fliiiix