ccMSC/ckb #41: Issue running driver daemon

ngleav, 2015-03-18:
Whenever i start the daemon, it causes my terminal window to constantly spam end of line characters. When i open another terminal window, it starts spamming new lines there as well. It is as if the driver is causing the enter key to be registered constantly.

Eventually, the other terminal starts erring out:

Attempting reset (17 of 25)...
Error: usbdequeue (firmware.c:23): Got return value 0xe00002cd
Attempting reset (18 of 25)...
Error: usbdequeue (firmware.c:23): Got return value 0xe00002cd
Attempting reset (19 of 25)...
Error: usbdequeue (firmware.c:23): Got return value 0xe00002cd
Attempting reset (20 of 25)...
Error: usbdequeue (firmware.c:23): Got return value 0xe00002cd
Attempting reset (21 of 25)...
Error: usbdequeue (firmware.c:23): Got return value 0xe00002cd
Attempting reset (22 of 25)...
Error: usbdequeue (firmware.c:23): Got return value 0xe00002cd
Attempting reset (23 of 25)...
Error: usbdequeue (firmware.c:23): Got return value 0xe00002cd
Attempting reset (24 of 25)...
Error: usbdequeue (firmware.c:23): Got return value 0xe00002cd
Attempting reset (25 of 25)...
Error: usbdequeue (firmware.c:23): Got return value 0xe00002cd

I have no idea whats wrong, Maybe someone here does? I used the pre packaged OSX app, i have not tried building it from source yet.

Thanks for your time.

tojuhaka, 2015-03-18:
This happens for me sometimes also. Just unplug the device and put it back.
Check the "Known issues" section in github.

ngleav, 2015-03-18:
Thanks, that fixed the issue. Now i'm trying to figure out how to change the color of all of the keys without animations.

speedfreak01, 2015-03-19:
You can set static colours very quickly in the CKB app. Create a profile, then select / highlight all keys on the keyboard displayed. Then select the colour you want from the colour pallet at the bottom left of the keyboard.

augustf, 2015-03-22:
I'm experiencing this issue under the latest 0.0.48-alpha when my BIOS switch is in the "1" position and neither plugging/unplugging nor restarting the daemon help. Even upon reboot, the issue continued, so I reverted to the 0.0.4.2 release without incident.

ccMSC, 2015-03-23:
See if v0.0.49 fixes it. If not, can you try the versions between 42 and 48 to see where it first appeared?

augustf, 2015-03-23:
@ccMSC Thanks! The key release code seemed to sort my issue. Making this a launch daemon was also a fantastic move.

ccMSC, 2015-03-24:
Awesome :D

Seems all the issues here have been sorted out, I'm going to go ahead and close this ticket.

closed 2015-03-24 by ccMSC