ccMSC/ckb #42: [Feature Request] Ability To Save Profiles to Software

y9yjRQcg7F, 2015-03-26:
At the moment, profiles will only save to hardware.  If the CKB frontend crashes for some reason, previous settings aren't retained and only the hardware saved profile will remain.

ccMSC, 2015-03-27:
Added in e96484b46cad85b6fcd3bbaf54ffa59977bc0e55 - I've extended the auto-save to occur once every 10 seconds if you've made any changes (previously, settings were only saved when quitting the app). Clicking the hardware save button will automatically save everything as well. Should be good enough to stop you from losing much if the app crashes.

closed 2015-03-27 by ccMSC

y9yjRQcg7F, 2015-03-28:
Thanks so much - you've added a feature and fixed a bug in 48 hours.  I'm sending you another PayPal tip - you're awesome!