ccMSC/ckb #43: [Bug] CKB Front-end Crashes Upon Restoring From System Tray

y9yjRQcg7F, 2015-03-26:
**OS:** Arch Linux x86_64
**Kernel:** 3.19.2-1-ARCH
**Windows Manager:** KDE Plasma 5
**Keyboard:** Corsair K70 RGB

The CKB daemon runs without issue and does not crash when the front-end does.  Here is a status systemd status check:

    # systemctl status ckb-daemon.service
    ● ckb-daemon.service - Corsair Keyboard Daemon
       Loaded: loaded (/usr/lib/systemd/system/ckb-daemon.service; enabled; vendor preset: disabled)
       Active: active (running) since Thu 2015-03-26 03:14:41 EDT; 1h 11min ago
     Main PID: 540 (ckb-daemon)
       CGroup: /system.slice/ckb-daemon.service
               └─540 /usr/bin/ckb-daemon

Here is a what posts when I run the CKB front-end:

    # ckb
    libpng warning: iCCP: known incorrect sRGB profile
    libpng warning: iCCP: known incorrect sRGB profile
    service is "org.kde.StatusNotifierItem-28413-1"
    Registering a client interface to the KStatusNotifierWatcher
    Scanning  "/opt/ckb/ckb-animations/ckb-gradient"
    Scanning  "/opt/ckb/ckb-animations/ckb-pinwheel"
    Scanning  "/opt/ckb/ckb-animations/ckb-rain"
    Scanning  "/opt/ckb/ckb-animations/ckb-random"
    Scanning  "/opt/ckb/ckb-animations/ckb-ripple"
    Scanning  "/opt/ckb/ckb-animations/ckb-wave"
    Starting  "/opt/ckb/ckb-animations/ckb-wave"
    Starting  "/opt/ckb/ckb-animations/ckb-gradient"

When I close the CKB front-end to the system tray, I get the following message:
>ckb will still run in the background.
>To close it, choose Exit from the tray menu
>or click "Quit ckb" on the Settings screen.

When I try to restore the front-end from the system tray the program crashes.  Here is what posts:

    service is "org.kde.StatusNotifierItem-28413-2"
    Registering a client interface to the KStatusNotifierWatcher

I'm not sure if this is a related issue, but when the front-end crashes, I lose all of the work I did previously - even if I've saved it to hardware.  At this point, I'm simply unable to create my own profiles due to this issue.

ccMSC, 2015-03-27:
I'm not experiencing the crash - can you run it in gdb and give me a backtrace? (`make debug && gdb bin/ckb`)

y9yjRQcg7F, 2015-03-28:
Here is the backtrace:

    (gdb) backtrace
    #0  0x00007ffff6dc8834 in QObject::thread() const () from /usr/lib/libQt5Core.so.5
    #1  0x00007ffff6dd3578 in QObject::QObject(QObject*) () from /usr/lib/libQt5Core.so.5
    #2  0x00007fffe323317e in DBusMenuExporter::DBusMenuExporter(QString const&, QMenu*, QDBusConnection const&) () from /usr/lib/libdbusmenu-qt5.so.2
    #3  0x00007fffea4d8585 in KStatusNotifierItem::setContextMenu(QMenu*) () from /usr/lib/libKF5Notifications.so.5
    #4  0x00007ffff79fe0ee in ?? () from /usr/lib/libQt5Widgets.so.5
    #5  0x00007ffff7a1ccde in ?? () from /usr/lib/libQt5Widgets.so.5
    #6  0x00007ffff79fdcc7 in QSystemTrayIcon::setVisible(bool) () from /usr/lib/libQt5Widgets.so.5
    #7  0x000000000041909f in QSystemTrayIcon::show (this=0x7f3590) at /usr/include/qt/QtWidgets/qsystemtrayicon.h:95
    #8  0x000000000041817e in MainWindow::showWindow (this=0x7fffffffe480) at mainwindow.cpp:218
    #9  0x00000000004a61bf in MainWindow::qt_static_metacall (_o=0x7fffffffe480, _c=QMetaObject::InvokeMetaMethod, _id=1, _a=0x7fffffffde50) at debug/moc_mainwindow.cpp:86
    #10 0x00007ffff6dca309 in QMetaObject::activate(QObject*, int, int, void**) () from /usr/lib/libQt5Core.so.5
    #11 0x00007ffff7676722 in QAction::triggered(bool) () from /usr/lib/libQt5Widgets.so.5
    #12 0x00007ffff7678d88 in QAction::activate(QAction::ActionEvent) () from /usr/lib/libQt5Widgets.so.5
    #13 0x00007ffff6dcb06a in QObject::event(QEvent*) () from /usr/lib/libQt5Core.so.5
    #14 0x00007ffff76792f1 in QAction::event(QEvent*) () from /usr/lib/libQt5Widgets.so.5
    #15 0x00007ffff7680acc in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /usr/lib/libQt5Widgets.so.5
    #16 0x00007ffff7685f10 in QApplication::notify(QObject*, QEvent*) () from /usr/lib/libQt5Widgets.so.5
    #17 0x00007ffff6d99d1b in QCoreApplication::notifyInternal(QObject*, QEvent*) () from /usr/lib/libQt5Core.so.5
    #18 0x00007ffff6d9bd5b in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) () from /usr/lib/libQt5Core.so.5
    #19 0x00007ffff6df1c83 in ?? () from /usr/lib/libQt5Core.so.5
    #20 0x00007ffff556f71d in g_main_context_dispatch () from /usr/lib/libglib-2.0.so.0
    #21 0x00007ffff556fa08 in ?? () from /usr/lib/libglib-2.0.so.0
    #22 0x00007ffff556fabc in g_main_context_iteration () from /usr/lib/libglib-2.0.so.0
    #23 0x00007ffff6df2077 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib    /libQt5Core.so.5
    #24 0x00007ffff6d977b2 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/libQt5Core.so.5
    #25 0x00007ffff6d9f18c in QCoreApplication::exec() () from /usr/lib/libQt5Core.so.5
    #26 0x0000000000415e79 in main (argc=1, argv=0x7fffffffe618) at main.cpp:42

ccMSC, 2015-03-28:
Looks like there's a bug caused by hiding and showing the tray icon rapidly. I did that as a workaround for an OSX problem anyhow, so I've removed those lines in the newest Linux version. See if it's working now.

y9yjRQcg7F, 2015-03-28:
This bug

    ( •_•)
    ( •_•)>⌐■-■
    (⌐■_■)

has been squashed.

ccMSC, 2015-03-29:
Awesome B)

closed 2015-03-29 by ccMSC