ccMSC/ckb #44: Intermittent key repeats

augustf, 2015-03-26:
Version: 0.0.4.9
Environment: OSX 10.10 (using ckb launch daemon)

After the issue I had relating to #41 went away some days ago, I've run into a much more intermittent problem (once in a full day of use). A key I'm pressing will begin repeating endlessly. I tried unplugging or just restarting ckb.app, but the only thing that works is to close ckb.app with my mouse, ssh into the machine from my laptop, unload the ckb launch daemon, reload it, and then start ckb.app.

This may be somewhat related to #41 (which occurred anytime I repeated a keystroke), but after encountering that problem, this is much more intermittent and the key involved seems to be random. Unlike the Known Issue regarding this, it's not connected with unplugging/replugging the keyboard; the key repeats continue without the keyboard.

ccMSC, 2015-03-27:
I'm wondering if this is caused by a hardware lockup? I've had a similar issue in Linux where the keyboard sometimes freezes after I press a key and the computer thinks it's disconnected (usually every few days). On Linux the key input automatically stops when the device disconnects, but I'm guessing OSX doesn't do that.

Anyway, see if a805c2d3ef99bddf7b226c0648ab98c79caafeb3 fixes it - at the very least it should stop if you unplug the keyboard now.

augustf, 2015-03-28:
I installed a805c2d earlier today, but a little while ago, when the key repeating began, unplugging the keyboard failed to stop it. Only intervening via SSH helped.

ccMSC, 2015-03-28:
Hmm. Well I'm stumped. I rewrote the OSX input system today, so it's possible I fixed it inadvertently, but I didn't notice anything that might cause this. Still, try it out and see if it's better now

augustf, 2015-03-28:
I'll build and run it today.

augustf, 2015-03-28:
Yikes-getting 2 build errors on my Mac. Is there a new dependency I don't have?

````
Undefined symbols for architecture x86_64:
  "_clock_gettime", referenced from:
      _os_keypress in input_mac.o
      _krthread in input_mac.o
      _main in main.o
  "_clock_nanosleep", referenced from:
      _main in main.o
  "_memrchr", referenced from:
      _readlines in devnode.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [../../../ckb-master/ckb.app/Contents/Resources/ckb-daemon] Error 1
make[1]: *** [debug] Error 2
make: *** [sub-src-ckb-daemon-make_first] Error 2
13:23:26: The process "/usr/bin/make" exited with code 2.
Error while building/deploying project ckb (kit: Desktop Qt 5.4.0 clang 64bit)
When executing step "Make"
````

ccMSC, 2015-03-29:
Whoops! I broke the Mac build when I fixed a compile error on Linux. It's fixed now.

augustf, 2015-04-03:
I wanted to try this for awhile before reporting back, and it's good news. I haven't had any issues with repeated keypresses since updating.

closed 2015-04-03 by augustf