ccMSC/ckb #45: auto install delays keypress animations

grosen, 2015-04-01:
After using the auto install functionality and copying the driver to the system folder keypress animations are being delayed. The delay seems to lessen when other keypress events are fired. Check out the video below. This has only started happening after I didn't run the daemon manually.

https://www.dropbox.com/s/lx13vouwpz2cfau/Video%20Apr%2001%2C%2011%2023%2024%20AM.mov?dl=0

ccMSC, 2015-04-02:
I think what happens is that launchd runs the process with a slightly lower priority than what it gets from running it in a Terminal, and that's why it has a slower response. I have plans to improve the latency in a future version.

grosen, 2015-04-06:
Great thanks! I'll monitor for any improvements.

grosen, 2015-04-27:
Seems to be working much faster now.

ccMSC, 2015-06-09:
As of v0.1.4 I've completed all the changes I had intended. Short list:
* ckb-daemon now runs with full resource limits regardless of whether started manually or from launchd
* The daemon will now accept RGB commands as soon as it's finished processing the last ones, whereas previously it ran on a timer which could cause some delay
* ckb will poll frames from animations more quickly when keys are pressed, enabling the animation to be ready immediately instead of delaying by one frame.

At this point I think the response time is as good as it can get, so I'm marking this issue as closed. Should be a noticeable improvement from the v0.1.3. :)

closed 2015-06-09 by ccMSC