ccMSC/ckb #47: Modifier keys not available in Steam games

augustf, 2015-04-03:
It may be other heavyweight fullscreen applications as well, but when I'm inside Steam, the ctrl, alt, and shift keys are unavailable. At first, I thought ckb had quit, but other keys that would not work if it had (like Escape and the number keys) work fine. I've no idea why it's just these keys/

ccMSC, 2015-04-03:
I wasn't able to replicate this issue (at least not in Source games) but it probably has to do with modifier keys having a different event type. See if 2979e17db82792493dcc1335105c05902744d904 fixes it.

fliiiix, 2015-04-03:
Which game?

augustf, 2015-04-03:
@fliiiix I was playing Grand Theft Auto Vice City on OS X Steam.

augustf, 2015-04-03:
The latest version worked for me -- all seems well.

closed 2015-04-03 by augustf