ccMSC/ckb #50: ckb --background opens gui

u83rmensch, 2015-04-10:
running fedora 21.     

I've tried adding the command to both the gui start up options and the rc.local script but get the same result when I boot. It works.. it boots ckb, but not quietly in the background it still launches the gui at boot.     

this isnt really a major issue necessarily because the fact that there are now any drivers at all is awesome but figure its probably something that should be brought up.

ccMSC, 2015-04-12:
Maybe it's getting run twice somehow? In 61181cab37b99132c67cac34582a3e809b0d316a I've changed it so that it won't open the GUI if that happens. See if that fixes your problem.

u83rmensch, 2015-04-12:
shouldnt be but certainly seems like it is. If I run >killall ckb then load it manually >ckb --background, it loads ckb quietly just fine, only place I have it loading though is in the rc.local file. How do I use that page to update my current version?

u83rmensch, 2015-04-13:
update: Think I figured out what the issue was. I dont think I installed your 61181ca right because running it twice still opened the gui. That aside, Fedora 21 was running it twice. I dont recall if this was a thing in Fedora KDE 20, but Fedora 21 w/ KDE by default has its session manager attempt to relaunch any programs and services  you had when you shut the computer down much like how a mac does. Im not fond of this practice and didnt realize it was doing it. That being said, it was launching it twice. After having removed the option to open programs and windows from previous session and only have > ckb --background in start up, it works as intended.

ccMSC, 2015-04-15:
You'd think it would be smart enough not to relaunch a program that's already running :P

Fair enough though, sounds like it's resolved.

closed 2015-04-15 by ccMSC