ccMSC/ckb #52: CKB app picks up wrong keyboard layout

Diserasta, 2015-04-15:
Hi there,
I'm running Arch Linux and the i3 wm.
When I boot up the app, it picks the wrong keyboard layout for me (with # next to return, and \ next to left shift). Now I've checked my locale (GB-UTF-8) and keyboard layout as given by systemd (us), and both should point to me having a standard layout keyboard (# is above 3 and \ is above return).
As an artifact of this, it will default the \ key to white as it can't find it to assign it a colour.
I'm going to try changing to a US locale when I get home (ISO or UTF-8 both) and see if that fixes it, but I have to run to class now.

ccMSC, 2015-04-16:
ckb detects layout based on the system locale - since yours says GB, it uses the GB layout. Changing the system locale to US should fix it, or else you can change the layout manually from the Device tab.

Diserasta, 2015-04-17:
Sweet. Thanks.

closed 2015-04-17 by Diserasta