ccMSC/ckb #53: Mac OS X- Menubar Icon Cannot be moved

c010rb1indusa, 2015-04-20:
The menubar icon for ckb.app cannot be moved with command+drag like normal menubar items. It doesn't show up in third party applications that manipulate menubar items either like 'Bartender'.

I'm running the software as a service.

ccMSC, 2015-04-21:
There's probably nothing I can do about it since the tray icon is a Qt class and I don't see any documentation about this. I'll keep the issue open in case anyone knows of a solution, though.

cocotutch, 2016-01-17:
Custom OS X Menulets cannot be moved. The Apple ones are a PrivateAPI and despite being an active OS X and iOS Developer for 10 years, have not yet found a public way to allow this behaviour.

This issue can be closed. :)

ccMSC, 2016-01-17:
Figured it was something like that. Shame. Thanks for sharing.

closed 2016-01-17 by ccMSC