ccMSC/ckb #54: default rainbow stops working..

u83rmensch, 2015-04-24:
I had this happen the first time I loaded ckb but I reinstalled my whole system (due to unrelated activities) and since reinstalling (and upgrading to fedora21 kde) Its been a pretty smooth ride.     

however this.. eh, I'll call it an inconvenience as its not really hindering me from using the keyboard, but the default Rainbow color scheme and animation seems to have discontinued working.    

This is a an open and close case though and thought I'd (continue) to share. In the middle of writing this I realized I did make one change. Under the "Settings" tab in ckb, i changed the FPS from 30 to 60. it seems the default rainbow animation doesnt like this and just turns all keys white with no animation. change it right back to 30 and it'll snap right back :D

(this may have also been the culprit for my weird key input lagging.. or maybe that was due to getting a bit of liquid in the keys while i was cleaning it.. idk)

ccMSC, 2015-04-25:
I *think* I know what caused this. I've had a fix in [testing] for a few days, I've merged it into [master] now. Let me know if you see the issue again after updating, otherwise I'll mark this as closed in a few days.

closed 2015-06-09 by ccMSC

ZhuofuChen, 2016-04-15:
Hello, I like your default animation, but I mistakely delete them, how can I get them back? I try to reinstall ckb application for many times, it still does not work