ccMSC/ckb #55: Failed to execute operation: Not a directory error during install

petterroea, 2015-04-26:
After building using quickinstall, it asks me if i want to install the driver as a startup device. Answering yes leaves me with a message showing "Failed to execute operation: Not a directory". After doing some debugging, i found the error to lie at line 135 in quickinstall:

```sudo systemctl enable ckb-daemon 2>$TMPFILE```

Runned on ubuntu 15.04

Terminal output:

```
System service: systemd detected
Run the driver as a startup service [Y/n]? y
Failed to execute operation: Not a directory

Press enter to exit.

```

ccMSC, 2015-04-26:
Looks like this was caused by `/usr/lib/systemd/system` not existing. 763e3a6362d46388ac8ee2203d0ca7581589635a should fix it.

Since the old script has created a file that isn't supposed to be there, you need to run
`sudo rm -f /usr/lib/systemd/system`
first, then try quickinstall again. After that everything should be fine.

closed 2015-06-09 by ccMSC

petterroea, 2015-06-11:
Yeah. I reinstalled due to some issues, so i couldn't test the fix, but i can confirm it did install as usual this time.

SNARKYPOTATO, 2017-01-17:
@ccMSC Please, please help me. I've spent the entire day trying to get the sodding thing to work and I'm just getting error after error. I have the K70 RGB LUX and it's amazing but I don't have any visual lights on it at all when on macOsierra.

petterroea, 2017-01-19:
@SNARKYPOTATO you need to make an issue with a detailed explanation of why things aren't working, and what you have allready tried, if you want any usefull help

mattanger, 2017-01-19:
@SNARKYPOTATO Sierra is pretty poorly supported right now. Also, you're not going to get any help from @ccMSC. We've got a new fork going that *might* work a little better with Sierra. https://github.com/mattanger/ckb-next/. That said, if you decide to try this fork, and you have problems that cannot be resolved by either reading the README or by searching through open or closed issue be sure to include a detailed description of your problem. That means, a description of what you've tried, and any relevant error messages.

mattanger, 2017-01-19:
@SNARKYPOTATO I see that you've already done this. Apologies.