ccMSC/ckb #56: ckb-daemon crash when loading ckb

penetal, 2015-04-27:
Version: beta-v0.1.2
Arch Linux: 3.19.3-3-ARCH #1 SMP PREEMPT Wed Apr 8 14:10:00 CEST 2015 x86_64 GNU/Linux

The daemon crashes when starting the interface.
It tells me that there is newer firmware available so I assume it reads some before it crashes.

## ckb-daemon load
The keyboard works, and is usable.

### ckb-daemon
> ckb Corsair Keyboard RGB driver beta-v0.1.2
> Setting FPS to 30
> Setting default layout: us
> Root controller ready at /dev/input/ckb0
> Connecting Corsair K95 RGB Gaming Keyboard (S/N: 1901303CAE3B940353759BAFF5001946)
> Device ready at /dev/input/ckb1

### dmesg
> [Apr27 06:23] input: ckb1 as /devices/virtual/input/input53

## ckb start
The interface shows up, but the keyboard will not be responsive as the daemon will now crash without any further output.

### dmesg
> [Apr27 06:24] traps: ckb-daemon[6406] general protection ip:7f7835308150 sp:7fffdcef3bd8 error:0 in libpthread-2.21.so[7f78352f6000+18000]

### ckb
> libpng warning: iCCP: known incorrect sRGB profile
> libpng warning: iCCP: known incorrect sRGB profile
> Scanning  "/opt/ckb/ckb-animations/ckb-gradient"
> Scanning  "/opt/ckb/ckb-animations/ckb-pinwheel"
> Scanning  "/opt/ckb/ckb-animations/ckb-rain"
> Scanning  "/opt/ckb/ckb-animations/ckb-random"
> Scanning  "/opt/ckb/ckb-animations/ckb-ripple"
> Scanning  "/opt/ckb/ckb-animations/ckb-wave"
> Starting  "/opt/ckb/ckb-animations/ckb-wave"
> Starting  "/opt/ckb/ckb-animations/ckb-gradient"
> Downloaded new firmware list. 3 entries found.

and then when quitting ckb I get this, though no processes are left running afterwards
> QProcess: Destroyed while process ("/opt/ckb/ckb-animations/ckb-wave") is still running.
> QProcess: Destroyed while process ("/opt/ckb/ckb-animations/ckb-gradient") is still running.
> QProcess: Destroyed while process ("sh") is still running.

I will be happy to help debug if you tell me what you need me to do.

ccMSC, 2015-04-27:
I found a double-unlock bug in the keyboard mutexes. Strangely, it wasn't causing a crash on my system? Anyway, should be fixed now - let me know if v0.1.3 is working for you

penetal, 2015-04-27:
# Fixed in v0.1.3

closed 2015-04-27 by penetal