ccMSC/ckb #57: Media Keys are not working at Mac Yosemite

yigitkonur, 2015-04-27:
I'm using latest drivers at CKB but forward and previous keys are not working. Are there any way to solve this problem? Thank you so much for your hard-working to port this great keyboard!

ccMSC, 2015-04-27:
Which media player are you using them with? Did they work without the driver?

yigitkonur, 2015-04-27:
It is not used with driver and non-driver mode. 

Just play/pause button is working. Stop-previous-next keys are not working.

notbrain, 2015-04-27:
I am using Karabiner to remap the media keys but have them on the Ins/Home/PgUp block for easier access, would be trivial to move them to the F9-F12 keys, with or without modifiers. ckb and Karabiner are overlapping in functionality here, but if you need to get it working Karabiner is a workaround for now. IMHO Karabiner has so much more with regard to keybindings it's kind of a distraction for ckb to provide the same (but I do wish Karabiner had a more ckb-like UI ;) ).

notbrain, 2015-04-28:
Just tried on my setup at home - the stop/play/seek keys are working for me on Yosemite 10.10.3 (14D136) and *iTunes*, with Fn key as modifier. Spotify only play/pause works, even with my remapped Karabiner seek keys. Spotify is not listening for the same codes.

yigitkonur, 2015-04-28:
Hey notbrain,

Am I doing something wrong in CKB configuration? You said that ckb-like UI but it is not useful and only have option for modifier keys which also supported by MacOS in its default.

[image]

yigitkonur, 2015-04-28:
Oh, I'm very sorry guys. I managed to use this after than re-install, driver is active now!

closed 2015-04-28 by yigitkonur

RobErskine, 2015-08-25:
I can also confirm that this is working fine on iTunes in osx 10.10. However, only play/pause are working within Spotify. @notbrain Did you find a way around this? Can Karabiner map these correctly?

k-sorrels, 2015-08-26:
Also can confirm the above.

ccMSC, 2015-08-27:
AFAIK this is due to the way that OSX handles media keys and there's no way around it. Maybe you could map the keys to something different in Spotify when the app has focus, but ckb doesn't support focus-based profile switching yet.

RobErskine, 2016-04-25:
I know this is closed now, but just incase anyone else is still having this issue, I was able to create a new profile and reassign the media keys and they now work properly. I was using a default profile before, not sure if that would mess with it or not.