ccMSC/ckb #59: Need support for international keyboard layout mapping

Rabcor, 2015-05-08:
I may have a US standard keyboard, but I need to be able to use the icelandic (is) mapping, I have my X11 set up to use icelandic keyboard layout, but it seems like the ckb-daemon is bypassing that layout.

Is there any way around this?

ccMSC, 2015-05-09:
At the moment the GUI only supports keys that are on the physical keyboard - support for additional keys is planned soon. I think you could simulate it by running `xdotool key` as a custom program, though.

Rabcor, 2015-05-12:
For now I'm solving it by setting it to another language in the GUI which does have this key (I think I'm using the swedish setting) seems to be working fine that way. although it perhaps isn't pretty.

ccMSC, 2015-07-24:
As of 9d2784dc8109724ce121d0cf497fc5ad9232d41f you should be able to assign the extra ISO key even with a US layout.

closed 2015-08-08 by ccMSC