ccMSC/ckb #60: CKB on mac crashes after restoring twice.

lukeberry99, 2015-05-12:
Happens each time. I open the app, hide it, click on restore. Hide it again, then when I try and restore the second time it crashes. I then have to open the app again twice to get it to work.

It's late, this might not make sense. I can make a video if you like?

P.S. Great work

ccMSC, 2015-05-12:
Works normally for me on Yosemite. Which version of ckb are you running?

lukeberry99, 2015-05-12:
Should note I'm using the precompiled binaries. alpha v0.0.52

ccMSC, 2015-05-13:
I think the problem was fixed in a more recent version. Try the latest binary (beta-v0.1.3):
https://github.com/ccMSC/ckb/releases/download/v0.1.3/ckb.pkg

lukeberry99, 2015-05-15:
That version of the client seems to be really unreliable, 90% of the time it crashes when attempting to save a profile. I'm going to build from source instead.

lukeberry99, 2015-05-15:
Not sure if I should start a second issue for this, as it is technically a different issue entirely.

Building from source off the master branch, the keyboard becomes unresponsive after any input that involves type lighting (for example, trippy and rainbow defaults). After pressing one key, the keyboard stops working and I have to unplug it and plug it back in.

This is the only application output I get.

```
Scanning  "/Users/luke/Desktop/ckb/ckb.app/Contents/Resources/ckb-animations/ckb-gradient"
Scanning  "/Users/luke/Desktop/ckb/ckb.app/Contents/Resources/ckb-animations/ckb-pinwheel"
Scanning  "/Users/luke/Desktop/ckb/ckb.app/Contents/Resources/ckb-animations/ckb-rain"
Scanning  "/Users/luke/Desktop/ckb/ckb.app/Contents/Resources/ckb-animations/ckb-random"
Scanning  "/Users/luke/Desktop/ckb/ckb.app/Contents/Resources/ckb-animations/ckb-ripple"
Scanning  "/Users/luke/Desktop/ckb/ckb.app/Contents/Resources/ckb-animations/ckb-wave"
qt.network.ssl: QSslSocket: cannot resolve TLSv1_1_client_method
qt.network.ssl: QSslSocket: cannot resolve TLSv1_2_client_method
qt.network.ssl: QSslSocket: cannot resolve TLSv1_1_server_method
qt.network.ssl: QSslSocket: cannot resolve TLSv1_2_server_method
qt.network.ssl: QSslSocket: cannot resolve SSL_select_next_proto
qt.network.ssl: QSslSocket: cannot resolve SSL_CTX_set_next_proto_select_cb
qt.network.ssl: QSslSocket: cannot resolve SSL_get0_next_proto_negotiated
qt.network.ssl: QSslSocket: cannot call unresolved function SSL_get0_next_proto_negotiated
Starting  "/Users/luke/Desktop/ckb/ckb.app/Contents/Resources/ckb-animations/ckb-gradient"
No GPG detected, signature verification disabled
Downloaded new firmware list. 3 entries found.
Starting  "/Users/luke/Desktop/ckb/ckb.app/Contents/Resources/ckb-animations/ckb-wave"
Starting  "/Users/luke/Desktop/ckb/ckb.app/Contents/Resources/ckb-animations/ckb-gradient"
Debugging has finished
```

ccMSC, 2015-05-15:
Try upgrading to the latest keyboard firmware (v1.20) from Windows.

ccMSC, 2015-05-20:
Fyi, I did some more testing and figured out that the v0.1.3 does indeed have problems dealing with the older firmware. If you can't/don't want to upgrade the FW for any reason, the latest version in the [testing](https://github.com/ccMSC/ckb/tree/testing) repo doesn't seem to have this issue.

lukeberry99, 2015-05-20:
Ahh cheers for doing the testing, I will upgrade the firmware when I get a chance - I feel its best to be on the latest version regardless - I'm able to do this via VMWare, yes?

I'm using the latest version from the testing repo at the moment, no issue there.

ccMSC, 2015-05-21:
You should be able to update the FW running Windows in a VM, yes. It's possible to update it from ckb as well, provided the device is working correctly in the first place. The safest option is to do it from an actual Windows PC, but that shouldn't be necessary unless you're paranoid.

closed 2015-06-09 by ccMSC