ccMSC/ckb #64: bind specific key to trigger animation

mhelvens, 2015-05-14:
Hi there. Thanks for ckb! It's working beautifully.

A low priority feature request: I'd like to bind a specific key to trigger an animation.

Attempt 1 at a clever work-around: I can have a specific 'mode' start out with a specific animation, and I can bind a key to a mode-switch. But a 'mode-switch' to the currently active mode does nothing. Sensible, really, in most cases.

Attempt 2 at a clever work-around: Two identical modes that switch to each other. Success! :-)

So, like I said, low priority. But it might be an easy job.

Cheers!

ruler501, 2015-09-26:
I might look into implementing this but am unsure of the best way to integrate it into the existing GUI.  I think it shouldn't take too long once that's figured out.

ccMSC, 2015-09-27:
So I've been working on this but I just realized I may have misunderstood the intent...In 37f947b4774af2f8753015f76ee50627358be8b2 there's now an option to start an animation in the center of the keyboard whenever you press a key. It repeats itself if you press any of the keys again, I'm not sure if that's what OP wanted or not. Anyway, it's a start.

ccMSC, 2015-10-14:
Ok, I was thinking CUE behaved differently for some reason but after checking it, I verified that the option I implemented was in fact the same thing that CUE does.

I wasn't quite happy with that option, though, so in 3c1ce6d5dba4825f8224d5ec52940c6847eefe22 I revised it to include the ability to start the animation only on the first keypress, but not on subsequent ones.

Lastly, in 75f1b31611e61e7a9126a1adc2b45ca44bf4b27a I've added the ability to trigger an animation from any key (via the Binding tab) even if it's not part of the animation. I'm fairly certain at least one of these should cover everyone's use cases, so unless somebody here can think of another scenario, this is probably going to be the final update for this option.

commit c546db94 (2015-11-11): ckb beta v0.2.2

Closes #64, #120, #133, #142, and #146.
closed 2015-11-11 by ccMSC