ccMSC/ckb #65: ckb-daemon loops on os_usbsend failures on OS X

cgaspar, 2015-05-14:
Starting ckb-daemon testing on OS X 10.10.3, I get the following output:

[I] Connecting Corsair K95 RGB Gaming Keyboard (S/N: 1703A037AE3B9403536DAE7DF5001942)
[I] Starting input thread for /var/run/ckb1
[E] os_usbsend (via led_keyboard.c:158): Got return value 0xe000404f
[I] Attempting reset...
[E] os_usbsend (via firmware.c:14): Got return value 0xe00002ed
[E] os_usbsend (via profile.c:308): Got return value 0xe000404f
[E] os_usbsend (via firmware.c:14): Got return value 0xe00002ed
[E] os_usbsend (via profile.c:308): Got return value 0xe000404f
[ repeat forever ]

master looks the same except for the text:

Connecting Corsair K95 RGB Gaming Keyboard (S/N: 1703A037AE3B9403536DAE7DF5001942)
usbdequeue (led.c:165): Got return value 0xe000404f
Attempting reset...
usbdequeue (firmware.c:21): Got return value 0xe00002ed
usbdequeue (firmware.c:21): Got return value 0xe000404f
usbdequeue (firmware.c:21): Got return value 0xe00002ed
[ repeat forever ]

cgaspar, 2015-05-14:
This works as expected when not connected via a USB hub, so something is unhappy traversing the hub :-(

ccMSC, 2015-05-15:
Can't find much info on what those errors actually mean, sadly (`2ed` is `kIOReturnNotResponding`, no idea what `404f` is though). What kind of USB hub is it (powered, unpowered, usb2, usb3, etc)?

cgaspar, 2015-05-16:
Powered USB 2 hub

ccMSC, 2015-05-20:
This does seem to be a problem with some hubs, even for windows users. I don't think there's anything I can do about it unfortunately. Some people have had luck experimenting with different connections (both plugs vs one plug) or moving the keyboard to different ports.

closed 2015-05-20 by ccMSC