ccMSC/ckb #69: Using Dvorak through CKB

dabfalcon, 2015-06-01:
First off, thank you for taking the time to make CKB. I would like to use CKB to remap my keyboard as Dvorak, which cannot be done in CUE. I brought this to Corsair's attention before I switched to Linux and didn't get a useful response. Here's a video from that explaining my problem (https://youtu.be/1g8AK3a8hQk?t=14).

I see that there is already an option in CKB for using 'United States (Dvorak)' but choosing it didn't help me. Is this something you could put into CKB?

ccMSC, 2015-06-02:
You need to set your system's keyboard layout to Dvorak (ckb won't do this for you because it doesn't support all languages). Usually you can find it under System Settings > Keyboard.

dabfalcon, 2015-06-02:
Thanks for the quick response. I understand that can be done but what I'm wondering about is a little different. Short story is I want to press a button on my keyboard, be typing in Dvorak, press another/the same button and be typing in QWERTY.

Long story: (I should specify, i'm looking at Programmer Dvorak. programmer-dvorak.appspot.com)
The most logical way I can think to do this is have the OS think the keyboard is one type, say QWERTY, all the time. Then use some of CKB's current functionality. Switch between M1 and M2. M1 would be standard QWERTY. M2 DVORAK. Secondly use the remap functionality to make 'f F' 'u U' and 'd D' 'e E'... This falls short when making '2 @' '@ ^', as explained in the video. Your software only allows key customisation providing the keys fall into standard QWERTY pairs.

Eg: QWERTY Hash:  Shift + key '#'      Key '3'
       DVORAK Hash: Key '#'      Shift + Key '`'

I hope this makes sense. I've tried to explain it very clearly. 

Thanks.

ccMSC, 2015-06-03:
Ahh I see the problem now. You can't make this work in ckb _per se_ because programmer Dvorak has different Shift+keys than the standard QWERTY/Dvorak layouts (at least not until I implement key combos, but that's several versions away).

It is possible to change the system layout with the command-line tool `setxkbmap`. I think it's installed by default, but try it in a terminal to make sure. To switch to the Dvorak layout:
`setxkbmap -layout us -variant dvp`
and to switch back to QWERTY:
`setxkbmap -layout us`
You can set ckb to run these using the "Program" option, which should do what you want. (note: use the default bindings in ckb since this will automatically remap them)

It's also possible to implement a toggle command on a single key:
```
bash -c "if [[ `setxkbmap -print | grep dvp | wc -l` -eq 0 ]]; then setxkbmap -layout us -variant dvp; else setxkbmap -layout us; fi"
```

dabfalcon, 2015-06-04:
Thanks ccMSC, that's awesome! I'll give it a shot one day soon. I'm also pleased to hear that my solution will probably be implemented way down the track. Having multiple ways to do something is often valuable.

dabfalcon, 2015-06-29:
Thanks again. I've used this a few times now :)

ccMSC, 2015-06-30:
Glad it's working!

closed 2015-06-30 by ccMSC