ccMSC/ckb #70: K95 (non RGB) stops working after reboot, OS X 10.10.3

conrad-vanl, 2015-06-01:
Not sure how to troubleshoot/debug. After a reboot can't get passed the login screen as the keyboard doesn't work, had to remove ckb from applications and manually remove the plist. Would be happy to re-install and attempt to troubleshoot/debug if there's anything I could do.

ccMSC, 2015-06-02:
Try replacing `/Library/LaunchDaemons/com.ckb.daemon.plist` with this:
```
<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
    <key>Label</key>
    <string>com.ckb.daemon</string>
    <key>ProgramArguments</key>
    <array>
        <string>/Applications/ckb.app/Contents/Resources/ckb-daemon</string>
        <string>--layout=us</string>
    </array>
    <key>KeepAlive</key>
    <true/>
    <key>StandardOutPath</key>
    <string>/var/log/ckb.log</string>
    <key>StandardErrorPath</key>
    <string>/var/log/ckb.log</string>
    <key>Debug</key>
    <true/>
</dict>
</plist>
```
It should write the daemon output to `/var/log/ckb.log`. Hopefully that has some clues about what's going on.

Two other things, may be related:
1. If you unplug and replug the keyboard, does it start working?
2. If, after logging in, you log out again, does the same thing happen? Or is it only when you first boot?

noxee, 2015-06-03:
I'm having a similar issue K70R.

I can get it working at the login screen by moving the polling switch on the back to another location.

I changed the plist file to start logging but nothing appeared in the log.

ccMSC, 2015-06-07:
I'm wondering if it's related to [this issue](http://forum.corsair.com/v3/showpost.php?p=782172&postcount=329).

I have a fix for it in the works now. If you're willing to compile the testing repository, you can try that out and see if it solves the problem.

noxee, 2015-06-07:
I've compiled and installed the testing repository and it looks like that has solved the problem.

Everything else seems to be working correctly as well. 

Thanks for that.

ccMSC, 2015-06-08:
Awesome, thanks. I'll merge the changes back to the master repo tomorrow (assuming I don't find any new problems).

Freurso, 2015-06-08:
I got a K65 RGB recently.But it does not work after reboot or every first boot(works fine after logging off) and it would take a while to get the keyboard recognized.So i am wondering if it is possible to fix that ?

ccMSC, 2015-06-09:
v0.1.4 is released - should solve this issue for everyone. Let me know if you're still having trouble after upgrading.

shahraj, 2015-06-11:
I have having an issue with 0.1.4 on OS X 10.10.3 where kb and the keyboard work on initial install from source. The binary I downloaded had M65 support and was buggy for some reason. But after disconnecting the keyboard or rebooting, ckb doesn't recognize the keyboard and it only works for a few second initially and then stops regardless of the position of the polling switch. Reverting back to 0.1.3 fixed all issues. I did a clean install of 0.1.4 as well after removing all traces of older versions of ckb to the same effect.

Freurso, 2015-06-11:
I got the same problem.Seems like something wrong with the driver.

ccMSC, 2015-06-12:
@shahraj @Freurso could you run the daemon (v0.1.4) manually and post the output? The command is
```
sudo launchctl unload /Library/LaunchDaemons/com.ckb.daemon.plist
sudo /Applications/ckb.app/Contents/Resources/ckb-daemon
```

shahraj, 2015-06-12:
I removed ckb and installed it via the quickinstall script, indicating yes to all the prompts. Then I ran the 2 commands you listed. Here is the output:

    ckb: Corsair RGB driver beta-v0.1.4
    [I] Root controller ready at /var/run/ckb0
    [I] Connecting Corsair K70 RGB Gaming Keyboard (S/N: REMOVED)
    [I] Starting input thread for /var/run/ckb1
    [E] os_usbrecv (via led_keyboard.c:163): Got return value 0xe0004051
    [I] Attempting reset...
    [E] os_usbrecv (via led_keyboard.c:163): Got return value 0xe0004051
    [E] os_usbrecv (via led_keyboard.c:163): Got return value 0xe0004051
    [E] os_usbrecv (via profile_keyboard.c:11): Got return value 0xe0004051
    [E] os_usbrecv (via led_keyboard.c:163): Got return value 0xe0004051
    [E] os_usbrecv (via led_keyboard.c:163): Got return value 0xe0004051
    [E] os_usbrecv (via led_keyboard.c:163): Got return value 0xe0004051
    [I] Reset success
    [I] Corsair K70 RGB Gaming Keyboard ready at /var/run/ckb1

Here is the output when I start the daemon manually then disconnect the keyboard then reconnect it. All seems fine in terms of use.

    ckb: Corsair RGB driver beta-v0.1.4
    [I] Root controller ready at /var/run/ckb0
    [I] Connecting Corsair K70 RGB Gaming Keyboard (S/N: REMOVED)
    [I] Starting input thread for /var/run/ckb1
    [E] os_usbrecv (via led_keyboard.c:163): Got return value 0xe0004051
    [I] Attempting reset...
    [E] os_usbrecv (via firmware.c:19): Got return value 0xe0004051
    [E] os_usbrecv (via led_keyboard.c:163): Got return value 0xe0004051
    [I] Reset success
    [I] Corsair K70 RGB Gaming Keyboard ready at /var/run/ckb1
    [E] os_usbsend (via led_keyboard.c:89): Got return value 0xe00002d7
    [I] Attempting reset...
    [E] os_usbsend (via firmware.c:14): Got return value 0xe00002d9
    [E] os_usbsend (via led_keyboard.c:89): Got return value 0xe00002d9
    [I] Reset success
    [E] os_usbsend (via led_keyboard.c:89): Got return value 0xe00002d9
    [I] Attempting reset...
    [E] usb_tryreset (usb.c:176): Reset failed. Disconnecting.
    [I] Disconnecting Corsair K70 RGB Gaming Keyboard (S/N: REMOVED)
    [I] Stopping input thread for /var/run/ckb1
    [I] Removed device path /var/run/ckb1
    [I] Connecting Corsair K70 RGB Gaming Keyboard (S/N: REMOVED)
    [I] Starting input thread for /var/run/ckb1
    [E] os_usbrecv (via profile_keyboard.c:47): Got return value 0xe0004051
    [I] Attempting reset...
    [E] os_usbrecv (via led_keyboard.c:163): Got return value 0xe0004051
    [E] os_usbrecv (via led_keyboard.c:163): Got return value 0xe0004051
    [E] os_usbrecv (via led_keyboard.c:163): Got return value 0xe0004051
    [E] os_usbrecv (via led_keyboard.c:163): Got return value 0xe0004051
    [E] os_usbrecv (via led_keyboard.c:163): Got return value 0xe0004051
    [I] Reset success
    [I] Corsair K70 RGB Gaming Keyboard ready at /var/run/ckb1

The issue starts after a reboot. Activity Monitor shows the daemon is running, but kb does not detect the kb nor does it work. Its simply lit up. Killing the daemon and restarting it appears to fix the issue. Below is the output for that.

    ckb: Corsair RGB driver beta-v0.1.4
    [I] Root controller ready at /var/run/ckb0
    [I] Connecting Corsair K70 RGB Gaming Keyboard (S/N: REMOVED)
    [I] Starting input thread for /var/run/ckb1
    [E] os_usbrecv (via led_keyboard.c:163): Got return value 0xe0004051
    [I] Attempting reset...
    [E] os_usbrecv (via firmware.c:19): Got return value 0xe0004051
    [E] os_usbrecv (via led_keyboard.c:163): Got return value 0xe0004051
    [E] os_usbrecv (via firmware.c:19): Got return value 0xe0004051
    [E] os_usbrecv (via led_keyboard.c:163): Got return value 0xe0004051
    [E] os_usbrecv (via led_keyboard.c:163): Got return value 0xe0004051
    [E] os_usbrecv (via led_keyboard.c:163): Got return value 0xe0004051
    [E] os_usbrecv (via profile_keyboard.c:35): Got return value 0xe0004051
    [E] os_usbrecv (via led_keyboard.c:163): Got return value 0xe0004051
    [I] Reset success
    [I] Corsair K70 RGB Gaming Keyboard ready at /var/run/ckb1

So it would seem there is some issue with the daemon at launch.

Freurso, 2015-06-12:
I ran the scripts for many times under different conditions.

When the KB totally failed to work

    ckb: Corsair RGB driver beta-v0.1.4
    [I] Root controller ready at /var/run/ckb0
    [E] iterate_devices (usb_mac.c:336): Failed to seize device: e00002c5
    [E] iterate_devices (usb_mac.c:336): Failed to seize device: e00002c5
    [E] iterate_devices (usb_mac.c:336): Failed to seize device: e00002c5
    [E] iterate_devices (usb_mac.c:336): Failed to seize device: e00002c5

After reinstalling with the .pkg file without rebooting.

    ckb: Corsair RGB driver beta-v0.1.4
    [I] Root controller ready at /var/run/ckb0
    [I] Connecting Corsair K65 RGB Gaming Keyboard (S/N: _CENCORED_)  
    [I] Starting input thread for /var/run/ckb1
    [E] os_usbrecv (via led_keyboard.c:163): Got return value 0xe0004051
    [I] Attempting reset...
    [I] Reset success
    [I] Corsair K65 RGB Gaming Keyboard ready at /var/run/ckb1

Reboot,and the KB failed to work, unplug and replug ,run the scripts and here comes the result

    ckb: Corsair RGB driver beta-v0.1.4
    [I] Root controller ready at /var/run/ckb0
    [I] Connecting Corsair K65 RGB Gaming Keyboard (S/N: 0F01D041AE3E148853B3E496F5001943)
    [I] Starting input thread for /var/run/ckb1
    [E] os_usbrecv (via led_keyboard.c:163): Got return value 0xe0004051
    [I] Attempting reset...
    [E] os_usbrecv (via led_keyboard.c:163): Got return value 0xe0004051
    [E] os_usbrecv (via led_keyboard.c:163): Got return value 0xe0004051
    [E] os_usbrecv (via led_keyboard.c:163): Got return value 0xe0004051
    [E] os_usbrecv (via profile_keyboard.c:11): Got return value 0xe0004051
    [E] os_usbrecv (via led_keyboard.c:163): Got return value 0xe0004051
    [I] Reset success
    [I] Corsair K65 RGB Gaming Keyboard ready at /var/run/ckb1

Hope this helpful for you.

ccMSC, 2015-06-13:
Something weird happened to the v0.1.4 binary that I uploaded. It was trying to connect the M65 RGB and was behaving strangely. Not sure how that happened, but I've replaced it with the correct binary now.

I noticed a handful of other bugs which might be causing this problem. There's a version with fixes applied here: https://github.com/ccMSC/ckb/tree/osxtest

If you need a binary package: https://dl.dropboxusercontent.com/u/16424004/ckb/v0.1.4%2Bosx1/ckb.pkg.zip

shahraj, 2015-06-13:
So upon initial install from source, the driver works great. Until reboot. Then kb does not recognizing the keyboard. Killing the daemon and restarting it manually with the commands unleashes all hell. The terminal output is: http://pastebin.com/1ZK6w3CU

Yes all the enters are in there. The keyboard just kept spamming enter and I couldn't even kill the process in terminal. I had to kill the daemon via Activity Monitor and then the keyboard was still spamming enters in the Terminal windows until I hit CNTRL+C to stop it.

ccMSC, 2015-06-14:
That's a known issue. You can fix it by unplugging the keyboard or sliding the poll rate switch, but it's best to start the daemon from a separate keyboard if possible.

I updated the osxtest tree with some small bugfixes. I also modified the service files so that you can get a log of what's happening at startup. Binary download: https://dl.dropboxusercontent.com/u/16424004/ckb/v0.1.4osx2/ckb.pkg

How to use:
- Install the new version (v0.1.4+osxtest2). If using quickinstall, set it to run as a service.
- In a terminal, run:
```
sudo launchctl unload /Library/LaunchDaemons/com.ckb.daemon.plist
sudo rm -f /var/log/ckb.log
```
- Reboot
- Open a new terminal and run
```
sudo launchctl unload /Library/LaunchDaemons/com.ckb.daemon.plist
```
- There should now be a complete log file at `/var/log/ckb.log`. To find it, use Cmd+Shift+G in Finder and go to `/var/log`.

It's normal to see some errors but it should eventually say the device is ready. Output from my system:
```
    ckb: Corsair RGB driver beta-v0.1.4+osx2
[I] Root controller ready at /var/run/ckb0
[I] Connecting Corsair K95 RGB Gaming Keyboard (S/N: *snip*)
[I] Starting input thread for /var/run/ckb1
[E] os_inputopen (input_mac.c:188): Unable to open IO service: 0xe00002cd
[I] Disconnecting Corsair K95 RGB Gaming Keyboard (S/N: *snip*)
[I] Removed device path /var/run/ckb1
[I] Stopping input thread for /var/run/ckb1
[I] Connecting Corsair K95 RGB Gaming Keyboard (S/N: *snip*)
[I] Starting input thread for /var/run/ckb1
[W] os_usbrecv (via profile_keyboard.c:28): Got return value 0xe0004051 (continuing)
[W] os_usbrecv (via led_keyboard.c:160): Got return value 0xe0004051 (continuing)
[W] os_usbrecv (via led_keyboard.c:160): Got return value 0xe0004051 (continuing)
[I] Corsair K95 RGB Gaming Keyboard ready at /var/run/ckb1

[I] Caught signal 15
[I] Disconnecting Corsair K95 RGB Gaming Keyboard (S/N: *snip*)
[I] Stopping input thread for /var/run/ckb1
[I] Removed device path /var/run/ckb1
[I] Removed device path /var/run/ckb0
```

shahraj, 2015-06-14:
My output is the same was what you see

    ckb: Corsair RGB driver beta-v0.1.4+osx2
   [I] Root controller ready at /var/run/ckb0
   [I] Connecting Corsair K70 RGB Gaming Keyboard (S/N: REMOVED)
   [I] Starting input thread for /var/run/ckb1
   [E] os_inputopen (input_mac.c:188): Unable to open IO service: 0xe00002cd
   [I] Disconnecting Corsair K70 RGB Gaming Keyboard (S/N: REMOVED)
   [I] Removed device path /var/run/ckb1
   [I] Stopping input thread for /var/run/ckb1
   [I] Connecting Corsair K70 RGB Gaming Keyboard (S/N: REMOVED)
   [I] Starting input thread for /var/run/ckb1
   [I] Corsair K70 RGB Gaming Keyboard ready at /var/run/ckb1
   [E] os_usbsend (via led_keyboard.c:89): Got return value 0xe000404f
   [I] Attempting reset...
   [E] os_usbsend (via device_keyboard.c:55): Got return value 0xe00002d9
   [E] usb_tryreset (usb.c:175): Reset failed. Disconnecting.
   [I] Disconnecting Corsair K70 RGB Gaming Keyboard (S/N: REMOVED)
   [I] Stopping input thread for /var/run/ckb1
   [I] Removed device path /var/run/ckb1
   [I] Connecting Corsair K70 RGB Gaming Keyboard (S/N: REMOVED)
   [I] Starting input thread for /var/run/ckb1
   [I] Corsair K70 RGB Gaming Keyboard ready at /var/run/ckb1

   [I] Caught signal 15
   [I] Disconnecting Corsair K70 RGB Gaming Keyboard (S/N: REMOVED)
   [I] Stopping input thread for /var/run/ckb1
   [I] Removed device path /var/run/ckb1
   [I] Removed device path /var/run/ckb0

Toggling the poll switch works fine.

Freurso, 2015-06-15:
My output
    
    ckb: Corsair RGB driver beta-v0.1.4+osx2
    [I] Root controller ready at /var/run/ckb0 
    [I] Connecting Corsair K65 RGB Gaming Keyboard (S/N: CENSORED)
    [E] os_inputopen (input_mac.c:188): Unable to open IO service: 0xe00002cd
    [I] Disconnecting Corsair K65 RGB Gaming Keyboard (S/N:CENSORED)
    [I] Starting input thread for /var/run/ckb1
    [I] Removed device path /var/run/ckb1
    [I] Stopping input thread for /var/run/ckb1
    [I] Caught signal 15
    [I] Removed device path /var/run/ckb0

Freurso, 2015-06-16:
I start up my Mac with the CKB 0.1.4(new) this morning ,but still can't make the KB work at login screen.But by  moving the polling switch, it works. The KB works fine after boot.

ccMSC, 2015-06-18:
I think the common issue here is `os_inputopen` failing with `kIOReturnNotOpen`. In @Freurso 's case it looks like the keyboard is disconnecting because of this. Not sure about @shahraj though, yours looks like it connected properly? Anyway, new version in the osxtest repo should eliminate the error. Let me know if it's any good - if not, follow the same instructions as before to get a log.

Binary: https://dl.dropboxusercontent.com/u/16424004/ckb/v0.1.4osx3/ckb.pkg.zip

My log:
```
    ckb: Corsair RGB driver beta-v0.1.4+osx3
[I] Root controller ready at /var/run/ckb0
[I] Connecting Corsair K95 RGB Gaming Keyboard (S/N: ...)
[I] Starting input thread for /var/run/ckb1
[E] os_usbsend (via led_keyboard.c:160): Got return value 0xe00002c2
[I] Attempting reset...
[E] usb_tryreset (usb.c:175): Reset failed. Disconnecting.
[I] Disconnecting Corsair K95 RGB Gaming Keyboard (S/N: ...)
[I] Stopping input thread for /var/run/ckb1
[I] Removed device path /var/run/ckb1
[I] Connecting Corsair K95 RGB Gaming Keyboard (S/N: ...)
[I] Starting input thread for /var/run/ckb1
[W] os_usbrecv (via profile_keyboard.c:28): Got return value 0xe0004051 (continuing)
[W] os_usbrecv (via led_keyboard.c:160): Got return value 0xe0004051 (continuing)
[W] os_usbrecv (via led_keyboard.c:160): Got return value 0xe0004051 (continuing)
[W] os_usbrecv (via led_keyboard.c:160): Got return value 0xe0004051 (continuing)
[W] os_usbrecv (via led_keyboard.c:160): Got return value 0xe0004051 (continuing)
[I] Corsair K95 RGB Gaming Keyboard ready at /var/run/ckb1

[I] Caught signal 15
[I] Disconnecting Corsair K95 RGB Gaming Keyboard (S/N: ...)
[I] Stopping input thread for /var/run/ckb1
[I] Removed device path /var/run/ckb1
[I] Removed device path /var/run/ckb0
```

Freurso, 2015-06-18:
@ccMSC the binary is AWESOME!!! It works now.The KB no more disconnecting itself at login.Thanks! Nice job.I may check if there is other issues.

ccMSC, 2015-06-19:
Nice!

Updated again, this time I've changed the profile loading so that it will give up after a certain number of failures and finish setting up anyway. Previous versions would loop indefinitely if they didn't succeed in loading the hardware profile.

I don't think anyone in this thread has had that problem, but I know that some other people do, so I'm including it in this branch.
https://dl.dropboxusercontent.com/u/16424004/ckb/v0.1.4osx4/ckb.pkg.zip

ccMSC, 2015-06-23:
All changes have been merged in v0.1.5. I'll leave this issue open for a while in case anyone is still experiencing it.

Freurso, 2015-06-24:
Thanks a lot

closed 2015-06-28 by ccMSC