ccMSC/ckb #71: Profiles/modes aren't saving

101Cipher010, 2015-06-07:
Hi, for some reason whenever I come up with a new mode and restart the computer or quit CKB the mode I had previously made does a rollback of sorts. As far as it has saved is the name of the profile and the name of the mode along with the color of the entire keyboard, but none of the animations.

I'm on Ubuntu 15.04, and I'm using a K70.

ccMSC, 2015-06-08:
Due to hardware limitations the animations won't work without ckb running. Do they come back when you restart the app?

If they're not being restored, look in `~/.config/ckb` and see if there's a file named `ckb.conf`. The settings should be stored there; if the file is missing or not being updated then that might be the problem.

101Cipher010, 2015-06-08:
They do not come back if ckb shuts down. I deleted the conf file and the same thing still happens.

ccMSC, 2015-06-08:
Does the conf file reappear after being deleted? Can you upload it somewhere (e.g. pastebin) and post it here?

101Cipher010, 2015-06-08:
http://pastebin.com/mu3wRhf7

ccMSC, 2015-06-08:
Well the animations are definitely saved there, so it seems they're not being loaded for some reason. I'll see if I can find out why.

101Cipher010, 2015-06-08:
Awesome, thanks

ccMSC, 2015-06-18:
I think the problem might have been ckb constantly re-loading the hardware profile and getting inconsistent results (it used to do a hardware save every time it exited, but that had its own problems).

There's a fix in [testing](https://github.com/ccMSC/ckb/tree/testing) now, assuming that this is the cause. If it's still not working, I'll put together a debugging build to figure out where the data is being lost.

101Cipher010, 2015-06-20:
Still having the same problem

ccMSC, 2015-07-28:
Hi, sorry it's been such a long time but I assume this was never fixed? I finally got that settings test build I was talking about. You can find it here: https://github.com/ccMSC/ckb/tree/settingtest

After installing, open it in a terminal with `/bin/ckb >~/ckb.log`. This should produce a file called `ckb.log` in your home directory with detailed information about where the settings are and how they're being loaded. Paste the output here and I'll try to figure out what's going on.

closed 2015-11-11 by ccMSC