ccMSC/ckb #72: K70 RGB Only works in the farthest bios switch to the left but CKB says no devices connected and sometimes it says driver innactive.

succession, 2015-06-16:
It was working earlier today and i made like 3 profiles but i left the rainbow effect running and went to go pick up my brother but when i came back (30 minutes later) it stopped working. I open CKB and it says "no devices connected" and sometimes  "driver innactive". Also when it was working i had it on switch "3" which is the second to last on the left. I tried unplugging the keyboard numerous times but nothing happened. I even plugged it into my windows laptop and it said i had the latest firmware. Im running Yosemite 10.10 and  on mac mid 2011. Please help im kind of worried now since i just bought this keyboard today too. Appreciate it, Thanks!

succession, 2015-06-16:
I FIXED IT!! YES. All i did was reinstall CKB! Wow this is great. The person who made this program is an amazing person.

ccMSC, 2015-06-17:
Thanks :D let me know if you have any other problems.

closed 2015-06-17 by ccMSC