ccMSC/ckb #74: Lighting Profiles

succession, 2015-06-17:
Is there anyway I can download some profiles off the internet and make them compatible with the k70 rgb through CKB on a Mac? Thanks!

ccMSC, 2015-06-18:
CUE profiles aren't supported at the moment. This is a planned feature, but still several versions down.

ruler501, 2015-09-26:
If someone has documentation on the file format for them I can look and see how hard importing(and maybe exporting)  them would be. I don't really have the time to figure out the format myself though. It seems to be XML but I'm not sure how to parse what look like GUID strings in it.

ccMSC, 2015-09-27:
I did a basic survey of the files and it looks like every mode, binding, animation, action, etc has its own GUID attached to it. So each of those things has an ID, and then any other element which references it will use that ID. They're a bit convoluted, tbh. I didn't notice any real magic numbers though, so it seems like it's just a matter of piecing the different sections together.

shaitand, 2015-12-10:
This would definitely be an awesome feature to have. I'm not really sure why it would be low on the list. It's nice to be able to manually try to come up with ideas but to be honest doing more than setting basic colors on things is a bit convoluted (not because the interface is bad really, there is just a lot that can be done). Having more animations and being able to utilize the work of others via cue profiles would be a huge plus.

shaitand, 2015-12-10:
@ccMSC - It sounds like you don't see this as a hugely difficult process. Although I know there are some compatibility issues between versions of the cue profiles for different devices. I found a compatibility list somewhere. Since this is important to me (and more importantly my wife) but I don't have time to learn the project and do it myself. I'll offer a bounty of 150USD to get import/export of community cue profiles working for at least the R70 RGB on linux mint (ckb currently working just fine) within the next month.

ccMSC, 2015-12-11:
@shaitand I appreciate the offer. It's not as low on the priority list as it used to be, but here's why I haven't worked on it yet:

1. ckb currently doesn't have feature parity with CUE in terms of binding or macro functionality. Given the amount of testing that's going to be required for this feature, I'd rather not have to modify it down the line. However, in light of your offer, I'll reconsider this.
2. ckb doesn't have the ability to import/export its own profiles yet, let alone CUE's. This needs to be addressed first.
3. I'm currently troubleshooting a number of bugs and performance issues, which has made progress on new features fairly slow.

If you're only interested in the lighting functionality and not the macros, I'll work on implementing this feature as soon as 2 and 3 are addressed.

shaitand, 2015-12-11:
@ccMSC Yes, I am primarily interested in the lighting functionality and not so much the macros (which don't impact us with macro key free K70's. I'm glad to hear you are willing to work on it. If possible I'd ask that this be done in such a way that if incompatible bindings/macros are in a cue profile it doesn't trigger a hard stop that prevents loading the lighting/animated functionality.

Who knows, maybe implementing just the lighting will make enough of the structure clear that stubs, placeholders, abstractions can be placed and set a skeleton framework that saves rather than takes more effort when it does come time to implement the rest of the features. Perhaps at some point ckb profiles and cue profiles could be stored in the same format.