ccMSC/ckb #76: ckb GUI doesn't launch

edtheuniquegeek, 2015-06-21:
Not sure what happened but after a reboot of my computer the GUI no longer launches. However, all configurations remain working. I even attempted to run the 'quickinstall' again and rebooted and it still does not launch. For now, all seems to be working on my keyboard. My custom colors and animation still function but someday I would like to be able to make changes...maybe.
Also, on a side not, and may need to be reported in a different issue thread, but after boot my computer seems to go into a sleep mode instead of the login screen. I have to wait a few seconds and start pressing buttons on the keyboard for the computer and monitors to come out of sleep mode before I see the login screen.
I'm using the K65 RGB in Ubuntu Mate 15.10.
By the way, nice job on this utility. Keep up the good work. Donation sent.

ccMSC, 2015-06-23:
The tray icon doesn't work on several environments (MATE being one of them) due to a Qt bug. You should be able to restore the GUI by relaunching ckb from the Applications menu.

t3kg33k, 2015-06-23:
Yeah, I'm not talking about the tray. I'm talking about the application launcher in the menu. It no longer launches when selected.

ccMSC, 2015-06-24:
Try killing the application and then restarting it (you may have to relaunch it twice).

In terminal:
```
$ killall ckb
$ ckb
ckb is already running. Exiting.
$ ckb
```

t3kg33k, 2015-06-24:
Glorious. That worked. Thanks.

ccMSC, 2015-06-24:
Awesome.

As a permanent fix, you may need to add a small delay when starting up the GUI. You can find the startup settings under `Control Center > Startup Applications > Startup Programs`. Edit the entry for ckb and change the "Command" field from `ckb --background` to `bash -c "sleep 5 && ckb --background"`. It's probably safe to use a smaller value like `2` but `5` is a good start. This should allow it to work normally across future reboots.

closed 2015-06-28 by ccMSC