ccMSC/ckb #77: Raindrop animation broken after release of v0.1.4

GigabyteProductions, 2015-06-23:
I've bisected and found that the first commit after v0.1.3 causes this issue.
```
$ git bisect log
# bad: [63a03bb96cd0ccc243b96584d32579be25acd494] v0.1.4 beta release
# good: [fa8b5dc92ae9fb3653454727e4459e19a09f95c2] ckb beta v0.1.3
git bisect start 'v0.1.4' 'v0.1.3'
# bad: [9a5831f88ac3a508bedfbf7995e3662501f8687e] (ckb-daemon) Removed global USB mutex (ckb-daemon) FPS command is back, replacing usbdelay command from last commit As I suspected, the multi-device problems were actually the result of the readlines bug, not a USB problem. Replaced `usbdelay` with `fps` because `fps` is more intuitive.
git bisect bad 9a5831f88ac3a508bedfbf7995e3662501f8687e
# bad: [ffd084827aa578f738ee4e9202c03c2d57f079fd] Version bump
git bisect bad ffd084827aa578f738ee4e9202c03c2d57f079fd
# bad: [cb268d9f46f2e0c276280382782c74f5b2a2d034] Version bump
git bisect bad cb268d9f46f2e0c276280382782c74f5b2a2d034
# bad: [498d1e5c9d0a7b15b0ceaa4d7a7dca3f5e51378c] (ckb) "Show animated" won't show dimming anymore
git bisect bad 498d1e5c9d0a7b15b0ceaa4d7a7dca3f5e51378c
# bad: [e12af6f70843d556ad6d09908207b56bafce53e0] (ckb) Option to display animated colors on-screen Enabled by default. Might cause a performance hit if your CPU isn't very good. (ckb) Small animation fixes
git bisect bad e12af6f70843d556ad6d09908207b56bafce53e0
# first bad commit: [e12af6f70843d556ad6d09908207b56bafce53e0] (ckb) Option to display animated colors on-screen Enabled by default. Might cause a performance hit if your CPU isn't very good. (ckb) Small animation fixes
```
The actual issue in question is that when the raindrop animation is set to start on key press, it only animates the first frame, and never resets (unless you tell it to reset after a set amount of time in the timings tab).

GigabyteProductions, 2015-06-23:
Maybe I did this wrong, or maybe I don't fully understand bisecting, because I see other commits right after v0.1.3 release...

GigabyteProductions, 2015-06-23:
Well I'm still confused with git's bisecting logic, but it obviously works, as I've confirmed that e12af6f70843d556ad6d09908207b56bafce53e0 does indeed cause this issue, because the commit immediately before it, 2544bcfeb4cffe1d5854251981ec89eceaed7285, is not subject to this issue.

ccMSC, 2015-06-24:
Fixed in testing: 53c6f02531a6e6ccce441226bcdb95da49ba15b7 (also added some comments so that the code is more readable...) Looks like when I split up the frame/time functions I inadvertently disabled the drops when "start with mode" wasn't enabled.

closed 2015-08-08 by ccMSC