ccMSC/ckb #80: Notification support

mrueg, 2015-06-26:
KDE Plasma (and I guess other DE as well) allows you to run a command, if a notification arrives at the notification server.
Having a "ckb-notify" application that is called by a notification event, reads the current keyboard color scheme and adds a fade-in-fade-out effect (like the notification LED on your smartphone) to a predefined key, would be great.

skeate, 2015-06-26:
This seems like it's probably outside the scope of what ckb should handle itself. As it is, it would be pretty easy to write a script to achieve this (by piping commands to `/dev/input/ckb1/cmd`)

ccMSC, 2015-06-27:
I have something sort of like this planned. What I want is a full command-line interface for ckb so that you can change profiles, start/stop animations, change key colors, etc from the command line (or, by extension, from shell scripts or anything that supports running a command). I'm not sure if that alone would accomplish what you want but it would at least be a solid framework.

SicVolo, 2016-01-09:
I am working on a libnotify animation plugin - I think this way it would be much easier for the ckb end user to setup, rather than a DE CLI catcher.

closed 2020-03-15 by mrueg