ccMSC/ckb #83: Pinwheel + new animation button - K70 new logo

WickedPadfoot, 2015-06-27:
Hi, I updated this morning for the 1.5 version and it look like the Pinwheel animation is not working anymore. Also, i can't click on the "new animation" button, it close my GUI but the application is still running. 

Thanks for you great job!
**I tried the windows driver lately and i can't believe how crappy it is compare to your!

ccMSC, 2015-06-28:
OSX or Linux? Have you tried rebooting and/or reinstalling it?

WickedPadfoot, 2015-06-28:
I'm on Linux. I have reboot and recompiled it.

ccMSC, 2015-06-30:
Strange, I'm not able to replicate the problem. You could try compiling [the testing repository](https://github.com/ccMSC/ckb/tree/testing) and see if that helps.

If not, could you send me your `ckb.conf` file? You can find it in `~/.config/ckb/`

WickedPadfoot, 2015-07-03:
My problem is now solved. I checked out the testing branch but the driver was still the same. I reinstalled it with quickinstall and now it's fine. You did a greath job on that update. Thank you!

closed 2015-07-03 by WickedPadfoot