ccMSC/ckb #85: GUI does not open on Ubuntu 15.04

bkloppenborg, 2015-07-09:
Recently (within the last month) the CKB GUI has failed to launch on my Ubuntu 15.04 system. I recall that `ckb` functioned correctly on my Ubuntu 14.04 system and also for some time on my 15.04 box, but evidently something changed which caused the GUI to stop working entirely. I have tried the workaround listed in #76; however, even after killing and restarting the process (via. a terminal) the GUI still will not appear. I thought this might be an issue with Gnome3; however, the `ckb` GUI fails to appear on Unity, KDE, and Gnome 3 so I suspect the problem lies somewhere else.

If I attach to `ckb-daemon` it appears to be executing normally as when I attach to the process my keyboard becomes unresponsive (as I would expect):

```
(gdb) where
#0  0x00007f52e1ef2743 in select () at ../sysdeps/unix/syscall-template.S:81
#1  0x0000000000409ea3 in usbmain () at usb_linux.c:383
#2  0x0000000000406f6d in main (argc=1, argv=0x7ffdc23e4ae8) at main.c:143

(gdb) info threads
  Id   Target Id         Frame 
  4    Thread 0x7f52e15f3700 (LWP 13559) "ckb-daemon" 0x00007f52e1ef20b7 in ioctl () at ../sysdeps/unix/syscall-template.S:81
  3    Thread 0x7f52e1df4700 (LWP 13571) "ckb-daemon" 0x00007f52e1ebfc4d in nanosleep () at ../sysdeps/unix/syscall-template.S:81
  2    Thread 0x7f52e0df2700 (LWP 13573) "ckb-daemon" 0x00007f52e1ef20b7 in ioctl () at ../sysdeps/unix/syscall-template.S:81
* 1    Thread 0x7f52e27dd740 (LWP 902) "ckb-daemon" 0x00007f52e1ef2743 in select () at ../sysdeps/unix/syscall-template.S:81
(gdb) continue
```

Interestingly every time I have attached to the `ckb` process with `gdb` it appears the process is getting stuck on updating the value in the `on_fpsBox_valueChanged` function. Making the function immediately return lets `ckb` advance further, but it hangs after loading the animations.

Are there any advanced debugging flags in `ckb` which I might enable to track down this issue? If not, do you have any suggestions on where I should look to find this perplexing problem?

ccMSC, 2015-07-09:
Can you give me the full stack traces from when it hangs on `on_fpsBox_valueChanged`, and when it hangs if you disable that? There might be some common function there that's causing trouble.

bkloppenborg, 2015-07-19:
Hi @ccMSC , sorry for the delay. I have a clean checkout of the current git master (81e3044f6e2d087dccd0c9bb915b0fffe624383a) and have cleaned and rebuilt from source. Below I have two full stacktraces for all threads for two cases: (1) an unmodified version of `ckb` which does not open a the GUI due to something getting stuck in the `on_fpsBox_valueChanged` function and (2) another full stacktrace where I have forced `on_fpsBox_valueChanged` to return immediately (the GUI still does not show up). Please let me know how else I might assist you in debugging this.

Case 1:

Launch `ckb`:
```
bkloppenborg@speedbird:~/workspace/ckb$ ./bin/ckb
 [no output on console]
```
Wait a few minutes, no GUI, attach using `gdb` running in a separate terminal:

```
(gdb) info threads
  Id   Target Id         Frame 
  4    Thread 0x7fe67168d700 (LWP 13187) "QXcbEventReader" 0x00007fe67c6228dd in poll ()
    at ../sysdeps/unix/syscall-template.S:81
  3    Thread 0x7fe66304f700 (LWP 13188) "gdbus" 0x00007fe67c6228dd in poll ()
    at ../sysdeps/unix/syscall-template.S:81
  2    Thread 0x7fe66284e700 (LWP 13189) "gmain" 0x00007fe67c6228dd in poll ()
    at ../sysdeps/unix/syscall-template.S:81
* 1    Thread 0x7fe67e711780 (LWP 13186) "ckb" 0x00007fe67d12e53d in nanosleep ()
    at ../sysdeps/unix/syscall-template.S:81
    
(gdb) thread 1
[Switching to thread 1 (Thread 0x7fe67e711780 (LWP 13186))]
#0  0x00007fe67d12e53d in nanosleep () at ../sysdeps/unix/syscall-template.S:81
81	in ../sysdeps/unix/syscall-template.S
(gdb) bt
#0  0x00007fe67d12e53d in nanosleep () at ../sysdeps/unix/syscall-template.S:81
#1  0x00007fe67d495d2d in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#2  0x00007fe67d3deef8 in QThread::msleep(unsigned long) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#3  0x00007fe67d5416c0 in QLockFile::tryLock(int) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#4  0x00007fe67d58db91 in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#5  0x00007fe67d58ee39 in QSettings::~QSettings() () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#6  0x00000000004b372c in CkbSettings::~CkbSettings (this=0x7fff9db234a0, __in_chrg=<optimized out>)
    at ckbsettings.cpp:12
#7  0x0000000000428cec in SettingsWidget::on_fpsBox_valueChanged (this=0x1bfe8e0, framerate=15)
    at settingswidget.cpp:168
#8  0x00000000004b7fc4 in SettingsWidget::qt_static_metacall (_o=0x1bfe8e0, 
    _c=QMetaObject::InvokeMetaMethod, _id=1, _a=0x7fff9db23670) at debug/moc_settingswidget.cpp:113
#9  0x00000000004b81f4 in SettingsWidget::qt_metacall (this=0x1bfe8e0, _c=QMetaObject::InvokeMetaMethod, 
    _id=1, _a=0x7fff9db23670) at debug/moc_settingswidget.cpp:155
#10 0x00007fe67d64eae2 in QMetaObject::activate(QObject*, int, int, void**) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#11 0x00007fe67e2da17e in QSpinBox::valueChanged(int) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#12 0x00007fe67e0beba0 in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#13 0x00007fe67e02acfa in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#14 0x00007fe67e0bed54 in QSpinBox::setValue(int) () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#15 0x0000000000427bd6 in SettingsWidget::SettingsWidget (this=0x1bfe8e0, parent=0x7fff9db23c70)
    at settingswidget.cpp:73
#16 0x00000000004172a3 in MainWindow::MainWindow (this=0x7fff9db23c70, parent=0x0) at mainwindow.cpp:68
#17 0x0000000000416832 in main (argc=1, argv=0x7fff9db23dd8) at main.cpp:46

(gdb) thread 2
[Switching to thread 2 (Thread 0x7fe66284e700 (LWP 13189))]
#0  0x00007fe67c6228dd in poll () at ../sysdeps/unix/syscall-template.S:81
81	../sysdeps/unix/syscall-template.S: No such file or directory.
(gdb) bt
#0  0x00007fe67c6228dd in poll () at ../sysdeps/unix/syscall-template.S:81
#1  0x00007fe67c00febc in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#2  0x00007fe67c00ffcc in g_main_context_iteration () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#3  0x00007fe67c010009 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#4  0x00007fe67c036955 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#5  0x00007fe67d1256aa in start_thread (arg=0x7fe66284e700) at pthread_create.c:333
#6  0x00007fe67c62deed in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109

(gdb) thread 3
[Switching to thread 3 (Thread 0x7fe66304f700 (LWP 13188))]
[... 133 lines cut ...]
81	in ../sysdeps/unix/syscall-template.S
(gdb) bt
#0  0x00007f62c3e0c8dd in poll () at ../sysdeps/unix/syscall-template.S:81
#1  0x00007f62c37f9ebc in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#2  0x00007f62c37fa242 in g_main_loop_run () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#3  0x00007f62b26bfaf6 in ?? () from /usr/lib/x86_64-linux-gnu/libgio-2.0.so.0
#4  0x00007f62c3820955 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#5  0x00007f62c490f6aa in start_thread (arg=0x7f62aa856700) at pthread_create.c:333
#6  0x00007f62c3e17eed in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109

gdb) thread 6
[Switching to thread 6 (Thread 0x7f62b8e77700 (LWP 13474))]
#0  0x00007f62c3e0c8dd in poll () at ../sysdeps/unix/syscall-template.S:81
81	in ../sysdeps/unix/syscall-template.S
(gdb) bt
#0  0x00007f62c3e0c8dd in poll () at ../sysdeps/unix/syscall-template.S:81
#1  0x00007f62c1a7bb72 in ?? () from /usr/lib/x86_64-linux-gnu/libxcb.so.1
#2  0x00007f62c1a7d64f in xcb_wait_for_event () from /usr/lib/x86_64-linux-gnu/libxcb.so.1
#3  0x00007f62bc0d6099 in ?? () from /usr/lib/x86_64-linux-gnu/qt5/plugins/platforms/libqxcb.so
#4  0x00007f62c4bc9b0e in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#5  0x00007f62c490f6aa in start_thread (arg=0x7f62b8e77700) at pthread_create.c:333
#6  0x00007f62c3e17eed in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109
```

In both cases it is hanging in the initial construction of the main window.

ccMSC, 2015-07-22:
It looks like the problem is with `QLockFile` somehow. Is `~/.config/ckb` missing or write-protected, or does it contain anything other than `ckb.conf`?

bkloppenborg, 2015-07-22:
The `~/.config/ckb` exists and has correct permissions:

```
drwxrwxr-x  2 bkloppenborg bkloppenborg  4096 Jun 19 08:38 ckb
```

However there are some files sitting around in the directory

```
drwx------ 41 bkloppenborg bkloppenborg  12288 Jul 20 21:40 ..
-rw-------  1 bkloppenborg bkloppenborg 157281 Jun 19 08:25 ckb.conf
-rw-------  1 bkloppenborg bkloppenborg 157281 Jun 19 08:38 ckb.conf.FW2187
-rw-r--r--  1 bkloppenborg bkloppenborg      0 Jun 19 09:45 ckb.conf.lock
``

Guessing that the presence of the lock file was messing up the initialization, I moved the '~/.config/cbk` directory and started up `ckb` and it successfully opened up the GUI. Given that this resolved the issue, is there any need for me to send you these files?

closed 2015-07-23 by bkloppenborg

ccMSC, 2015-07-23:
No, I think they just need to be removed. I can probably add a workaround for that - they're supposed to prevent multiple processes from accessing the file at the same time, but ckb already runs as a single instance, so it should be safe.