ccMSC/ckb #86: Feature request: color scheme based on caps lock (nums lock, scroll lock, etc)

conrad-vanl, 2015-07-09:
It would be really cool to be able to specify color schemes when in certain keyboard modes, such as caps lock. I'm thinking could have the "Caps Lock" key glowing a different color when caps lock is on.

ccMSC, 2015-07-10:
This is planned for v0.2 - it's very similar to the DPI indicator function on the mouse, there's going to be an option to change the color of the key instead of using the indicator lights. Stay tuned.

ccMSC, 2015-08-24:
So this *was* planned for v0.2 but then I scrapped it due to firmware issues. However, in c7bc89d2b43bff6ce22d97c281c374670d207fa3 it has been implemented. You have the option to use an RGB indicator instead of (or in addition to) the normal one, or the indicators can be turned off altogether.

closed 2015-08-26 by ccMSC