ccMSC/ckb #88: Timeline of m65 support

JackoPlane, 2015-07-10:
Any update on m65 support timelines?

ccMSC, 2015-07-12:
Most likely 2-3 weeks. The driver portion is finished, it just needs a GUI.

Freurso, 2015-07-17:
And……M95? Wanna buy one.

ccMSC, 2015-07-17:
M95 support is unlikely unless they make an RGB version. The non-RGB products use a completely different USB protocol, so it's not possible to add support for them unless someone else decodes the protocol (and even then, non-RGB support has been spotty, for the simple reason that I'm not able to test them myself).

Freurso, 2015-07-20:
It's OKay.Logitech MX master maybe an option for me.

JackoPlane, 2015-07-24:
Thanks for the update ccMSC, Hows the interface progressing?

ccMSC, 2015-07-24:
Just pushed a major improvement now :)

The main thing that's left to do is the performance/DPI tab. After that, a few small additions to the animation functionality, then it's just regression testing until v0.2 is released.

JackoPlane, 2015-07-24:
Thanks for that @ccMSC, I've built and installed the daemon, I can control RGB colours now, however clicks, and mouse movement aren't working. Is this a known issue?

JackoPlane, 2015-07-25:
Update - Daemon console log:

```
$ sudo /Applications/ckb.app/Contents/Resources/ckb-daemon
    ckb: Corsair RGB driver beta-v0.1.5+t09
[I] Root controller ready at /var/run/ckb0
[I] Connecting Corsair M65 RGB Gaming Mouse (S/N: 14008027AE3B940853589ABBF5001947)
[I] Starting input thread for /var/run/ckb1
[I] Corsair M65 RGB Gaming Mouse ready at /var/run/ckb1
[W] postevent (input_mac.c:44): Post event failed: e00002c2
[W] postevent (input_mac.c:44): Post event failed: e00002c2
[W] postevent (input_mac.c:44): Post event failed: e00002c2
[W] postevent (input_mac.c:44): Post event failed: e00002c2
[W] postevent (input_mac.c:44): Post event failed: e00002c2
[W] postevent (input_mac.c:44): Post event failed: e00002c2
[W] postevent (input_mac.c:44): Post event failed: e00002c2
[W] postevent (input_mac.c:44): Post event failed: e00002c2
[W] postevent (input_mac.c:44): Post event failed: e00002c2
^C
[I] Caught signal 2
[I] Disconnecting Corsair M65 RGB Gaming Mouse (S/N: 14008027AE3B940853589ABBF5001947)
[I] Stopping input thread for /var/run/ckb1
[I] Removed device path /var/run/ckb1
[I] Removed device path /var/run/ckb0
```

ccMSC, 2015-07-25:
Odd, I haven't seen that error before. Is this on Yosemite or Mavericks?

I pushed an update today to fix an unrelated problem - I don't think that's the cause of this issue, but it's worth a try.

JackoPlane, 2015-07-25:
@ccMSC This is on a El Capitan machine, Will try testing on a Yosemite machine.

ccMSC, 2015-07-28:
Strange, I tried installing the latest El Capitan beta and it's still working for me. Let me know if you have any luck.

JackoPlane, 2015-07-28:
Appears to have been an issue with beta 3. Danm beta OS's :)

closed 2015-07-28 by JackoPlane