ccMSC/ckb #89: Icon is in a wierd location

timeDev, 2015-07-12:
Using Ubuntu 14.04, the ckb icon appears right in the top left corner of the screen, where the x is located, unlike other applications like steam, which are in the top right.

ccMSC, 2015-07-14:
This is being worked on in the [testing repo](https://github.com/ccMSC/ckb/tree/testing). If you have `libappindicator` installed you can compile that version and it should appear correctly.

johnkernke, 2015-07-21:
I have a few different versions of `libappindicator` installed:

```
$ dpkg --get-selections | grep libappindicator
libappindicator1				install
libappindicator3-1				install
```

But the way that it checks if it is installed doesn't work.

```
$ pkg-config --exists appindicator-0.1
$ echo $?
1
```

I also tried installing the package `libappindicator0.1-cil-dev` but it also wouldn't get picked up with `pkg-config`.

Any recommendations on how to get it to be picked up, preferably installed through apt.

skandalfo, 2015-07-21:
libappindicator0.1-cil-dev seems to be the .NET devel package for libappindicator. I think you want just plain libappindicator-dev.

johnkernke, 2015-07-21:
ah, thanks. that was the package I needed.

closed 2015-08-08 by ccMSC