ccMSC/ckb #90: K95 non-RGB keypress issue

SimonWoodburyForget, 2015-07-14:
The issue is that when i press a key it then stops being pressed, between 0 - 15 seconds after usually.

So if i where to type a password say "PASSWORD!!!" it often comes out "PASsword111" or even "password111" where shift just stopped working right away, same goes for every other combination of keypress. If i was to hold a key (xxxxxxxxxxxxxxxxxx), it will stop recurring after 1 to 30 times.

Running: Linux Mint Cinnamon 17.2 64-bit

ccMSC, 2015-07-16:
Have you tried it without the driver? I don't have a non-RGB board to test but this sounds like it might be a hardware problem.

SimonWoodburyForget, 2015-07-16:
Yes it worked with the standard drivers it always worked, with my windows 7 system with the standard corsair drivers to, i even tried reinstalling the drivers, din't expect much but din't do anything.

Now i just tested it on my second system: Mint Cin 32-bit, (old different pc, fresh install) and it also has the same issue, well at less i know its the keyboard and nothing else pretty much.

ccMSC, 2015-07-17:
Might have found the problem. Try installing the [latest testing version](https://github.com/ccMSC/ckb/tree/testing) and see if it's fixed now.

SimonWoodburyForget, 2015-07-18:
Uninstalled the master version and installed your testing version beta-v0.1.5+t08. 

Getting the same issue.

ccMSC, 2015-07-22:
Ok, I'll try to get a debug version ready in a few days to figure out what's going on.

ccMSC, 2015-07-28:
Test branch is up: https://github.com/ccMSC/ckb/tree/inputtest

Install as usual. To use it, first stop the system service (it will be either `sudo systemctl stop ckb-daemon` or `sudo service ckb-daemon stop` depending on whether you've got systemd or not). Then open a terminal and run the driver with `sudo /bin/ckb-daemon`. It should print a bunch of information every time you press a key. Try to replicate the problem. When you're done, stop the driver by pressing Ctrl+C in the terminal window, then paste the output here.

SimonWoodburyForget, 2015-07-30:
Installed the driver and stopped the services ran the script script(it was not in /bin/ it was in /usr/) din't seem to do anything(i saw the ui but it just typed what i was typing, and not reproducing the error), restarted my computer stopped services again and ran the script and the error started happening again.

Output:

    simon@simon-GA-990FXA-UD3 ~ $ sudo /usr/bin/ckb-daemon 
        ckb: Corsair RGB driver beta-v0.1.5
    [I] Root controller ready at /dev/input/ckb0
    [I] Connecting Corsair K95W Gaming Keyboard (S/N: )
    [I] Starting input thread for /dev/input/ckb1
    [I] Corsair K95W Gaming Keyboard ready at /dev/input/ckb1
    SSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSSDSDAFDSFSADFGDFGDFS[E] _nk95cmd (via device_keyboard.c:138): Operation not permitted
    gfdsgsdasdfsdafsdafsdafsadfsdafsdafsdafsdafsdafSDFSDAFSDAFSDAFSDAF[E] _nk95cmd (via device_keyboard.c:138): Operation not permitted
    sdafsdafdsafasdfsdafsdafSADFASDFASDFSADFASDFSADFASDFSADFSADFASDFASDF[E] _nk95cmd (via device_keyboard.c:138): Operation not permitted
    asdfasdfASDFASDFASDFASDFASDFASDFASDFASDFASDFASDFASDFSADFASDFASDF[E] _nk95cmd (via device_keyboard.c:138): Operation not permitted
    asdfasdfsda[E] _nk95cmd (via device_keyboard.c:138): Operation not permitted
    [E] _nk95cmd (via device_keyboard.c:138): Operation not permitted
    [E] _nk95cmd (via device_keyboard.c:138): Operation not permitted
    [E] _nk95cmd (via device_keyboard.c:138): Operation not permitted
    [E] _nk95cmd (via device_keyboard.c:138): Operation not permitted
    [E] _nk95cmd (via device_keyboard.c:138): Operation not permitted
    [E] _nk95cmd (via device_keyboard.c:138): Operation not permitted
    [E] _nk95cmd (via device_keyboard.c:138): Operation not permitted
    [E] _nk95cmd (via device_keyboard.c:138): Operation not permitted
    fsadfsadfsdafasdfsadfASDFASDFASDFDAFGSDFAGSDFGSDFGASDFGDAFGSDFGSDFGSDFGSDFGSDFGSDF[E] _nk95cmd (via device_keyboard.c:138): Operation not permitted
    gsdfgsdfgsdfg[E] _nk95cmd (via device_keyboard.c:138): Operation not permitted
    ^C
    [I] Caught signal 2
    [E] _nk95cmd (via device_keyboard.c:138): Operation not permitted
    [I] Disconnecting Corsair K95W Gaming Keyboard (S/N: 1b1c:1b08-NoID)
    [I] Removed device path /dev/input/ckb1
    [I] Removed device path /dev/input/ckb0
    simon@simon-GA-990FXA-UD3 ~ $ 

So i just simply held shift while typing other keys, then that "operation not permitted" pops up almost when it cuts off(i tried a few more times, and . It'll just print every few seconds.

ccMSC, 2015-07-30:
I don't think that's the right version. It should say `beta-v0.1.5+t11` at the top. Did you run the quickinstall script from the `ckb-inputtest` folder like normal?

It's entirely possible the problem is caused by `nk95cmd`, though. I've never actually gotten confirmation that it works so I'll probably just disable it if I can't figure out what's going on.

SimonWoodburyForget, 2015-07-30:
Sorry i downloaded the zip file, and it seems like github decided to download the master even thought i was on the input(seems to be a weird browser bug, i had it happen last time i think) and i totally skimmed over checking which version i had.

Anyways i quick installed the +t11 version and tested it.

Output:

    simon@simon-GA-990FXA-UD3 ~ $ sudo /usr/bin/ckb-daemon
        ckb: Corsair RGB driver beta-v0.1.5+t11
    [I] Root controller ready at /dev/input/ckb0
    [I] Connecting Corsair K95W Gaming Keyboard (S/N: )
    [I] Starting input thread for /dev/input/ckb1
    [I] Corsair K95W Gaming Keyboard ready at /dev/input/ckb1
    Input 0x81 (0x08): 0000000000000000
    N-RGB switch
    kb: 1 (8)
    Input 0x81 (0x08): 0000000000000000
    N-RGB switch
    kb: 1 (8)
    Input 0x81 (0x08): 0000000000000000
    N-RGB switch
    kb: 1 (8)
    Input 0x81 (0x08): 0000000000000000
    N-RGB switch
    kb: 1 (8)
    Input 0x81 (0x08): 0000000000000000
    N-RGB switch
    kb: 1 (8)
    Input 0x81 (0x08): 0000000000000000
    N-RGB switch
    kb: 1 (8)
    Input 0x81 (0x08): 0000000000000000
    N-RGB switch
    kb: 1 (8)
    Input 0x81 (0x08): 0000000000000000
    N-RGB switch
    kb: 1 (8)
    Input 0x81 (0x08): 0000000000000000
    N-RGB switch
    kb: 1 (8)
    Input 0x81 (0x08): 0000000000000000
    N-RGB switch
    kb: 1 (8)
    Input 0x81 (0x08): 0000000000000000
    N-RGB switch
    kb: 1 (8)
    Input 0x81 (0x08): 0000000000000000
    N-RGB switch
    kb: 1 (8)
    Input 0x81 (0x08): 0000000000000000
    N-RGB switch
    kb: 1 (8)
    Input 0x81 (0x08): 0000000000000000
    N-RGB switch
    kb: 1 (8)
    Input 0x81 (0x08): 0000000000000000
    N-RGB switch
    kb: 1 (8)
    Input 0x81 (0x08): 0000000000000000
    N-RGB switch
    kb: 1 (8)
    Input 0x81 (0x08): 0000000000000000
    N-RGB switch
    kb: 1 (8)
    Input 0x81 (0x08): 0000000000000000
    N-RGB switch
    kb: 1 (8)
    Input 0x81 (0x08): 0000000000000000
    N-RGB switch
    kb: 1 (8)
    Input 0x81 (0x08): 0000000000000000
[... 173 lines cut ...]
    kb: 1 (8)
    Input 0x81 (0x08): 0000000000000000
    N-RGB switch
    kb: 1 (8)
    Input 0x81 (0x08): 0000000000000000
    N-RGB switch
    kb: 1 (8)
    Input 0x81 (0x08): 0000000000000000
    N-RGB switch
    kb: 1 (8)
    Input 0x81 (0x08): 0000000000000000
    N-RGB switch
    kb: 1 (8)
    Input 0x81 (0x08): 0000000000000000
    N-RGB switch
    kb: 1 (8)
    [I] Disconnecting Corsair K95W Gaming Keyboard (S/N: 1b1c:1b08-NoID)
    Input 0x81 (0x00): 
    N-RGB switch
    kb: 1 (0)
    [I] Removed device path /dev/input/ckb1
    [I] Removed device path /dev/input/ckb0
    simon@simon-GA-990FXA-UD3 ~ $ 

I just held the S letter with shift till it did the error and ctrl+c out.

ccMSC, 2015-08-05:
I think the problem was caused by the driver constantly spamming the keyboard with mode-switch messages. I've updated the inputtest branch (`beta-v0.1.6+t12`) so try it now and see if it's fixed.

SimonWoodburyForget, 2015-08-05:
Looks awesome, it fixed the bug. I set some macros and used some settings in your GUI, everything seems to be working normally. Thanks a lot!

closed 2015-08-08 by ccMSC