ccMSC/ckb #93: Firmware 1.30

fliiiix, 2015-07-29:
I upgraded to the Firmware 1.30 on my K70RGB. And this does not work for my with linux and ckb.
I downgraded to the Firmware 1.15 (works fine with ckb) since this is the only on I found if any one has the 1.20 file, it would be nice if you can share it!

Btw. crosair update the content of this links in
https://github.com/ccMSC/ckb/blob/master/FIRMWARE

> Corsair K70RGB 1.20 http://www3.corsair.com/software/HID/K70RGB.zip

is now a link to the 1.30 Firmware

If I can test anything or you need additional infos just ask.

Tazmain, 2015-07-29:
I am having issues with the 1.30 firmware and ckb as well. When the daemon is running I have a input delay and not all key strokes get recorded. As soon as I stop the daemon, everything is back to normal. (Try entering a password when every 2de or 3rd key doesn't get recorded)

Tazmain, 2015-07-29:
Here is my full dmesg as well. http://sprunge.us/TMaL

ccMSC, 2015-07-30:
Done! Updated driver to support new firmware and pushed changes to master as v0.1.6. Let me know if you have any difficulties.

Slightly related: I noticed that `K65RGB.zip` is gone from Corsair's servers so I removed it from ckb's firmware list. I don't know if Corsair have stopped supporting it or if the file's simply moved somewhere. If anyone knows where it went, I'll put it back.

fliiiix, 2015-07-30:
I just tried it does not work for me

```
Jul 30 11:26:55 localhost.localdomain ckb-daemon[2396]: [E] os_usbrecv (via firmware.c:15): No data read
Jul 30 11:26:56 localhost.localdomain ckb-daemon[2396]: [E] os_usbrecv (via firmware.c:15): No data read
Jul 30 11:26:56 localhost.localdomain ckb-daemon[2396]: [E] os_usbrecv (via firmware.c:15): No data read
Jul 30 11:26:57 localhost.localdomain ckb-daemon[2396]: [E] os_usbrecv (via firmware.c:15): No data read
Jul 30 11:27:02 localhost.localdomain ckb-daemon[2396]: [E] os_resetusb (via usb.c:167): usbunclaim failed: No such device
Jul 30 11:27:02 localhost.localdomain ckb-daemon[2396]: [E] usb_tryreset (usb.c:175): Reset failed. Disconnecting.
Jul 30 11:27:28 localhost.localdomain ckb-daemon[2396]: [E] os_setupusb (usb_linux.c:246): Failed to claim interface: No route to host
```

the `os_usbrecv (via firmware.c:15): No data read` repeats many times more

dassystem, 2015-07-30:
Doesn't work for me either with v0.1.6 on 1.30 firmware. @fliiiix can you share your 1.15 firmware? I cannot find it online.

Tazmain, 2015-07-30:
Same here v0.1.6 does not work on 1.30 firmware. Ckb app does not pick up a keyboard is plugged-in. Keyboard still laggs when the daemon is running. I built from git using the quickinstall.

fliiiix, 2015-07-30:
sure, but with 1.15 http://file.l33t.name/k70rgb_app115-bin it doesn't work in grub for some reason

ccMSC, 2015-07-31:
I think I messed something up when updating the master repo. For now try installing the testing version instead.

fliiiix, 2015-07-31:
Amazing, testing works for me! :+1: 

I would leave this bug open till it's merged into master.

closed 2015-08-08 by ccMSC

mrueg, 2015-08-08:
in FIRMWARE: 
 Corsair K70RGB 1.20 http://www3.corsair.com/software/HID/K70RGB.zip 0.1.6   K70RGB_app130.bin e4ec7852d0457e3eeea794f2a278228c3eb3e26de03973960afa453698968cf5
  Corsair K95RGB 1.20 http://www3.corsair.com/software/HID/K95RGB.zip 0.1.6   K95RGB_app130.bin 192d3ca125cb560dce2a764b9e29491875d5073c34b0845de48e805a27de3699

Why has the bin name changed to 130 but the firmware version is still at 1.20?

ccMSC, 2015-08-09:
oops...fixed. Also changed the min ckb version to 0.2 since 0.1.6 didn't really work.