ccMSC/ckb #96: Doesn't respond to keys

KraXarN, 2015-08-04:
Driver installed fine, it detects my Corsair K95 fine and animations works, but the keys won't actually work. No matter what I press, nothing happens, except for the media keys, which seems to work fine...
I have a swedish/nordic keyboard layout, dunno if that's the issue.
After disabling the ckb-daemon service, the keyboard starts working again.

elewis787, 2015-08-04:
Same, installed prefect but keys are not responding as they should.

ccMSC, 2015-08-05:
Which OS? Have you tried compiling [the testing branch](https://github.com/ccMSC/ckb/tree/testing)?

KraXarN, 2015-08-05:
Currently running Kubuntu 15.04 x64
3.19.0-25-generic #26-Ubuntu SMP Fri Jul 24 21:17:31 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux

Testing branch works fine though, thanks for making this awesome driver! :)

Only thing, I don't know if this is just because I'm on the testing branch, but the ckb program says a warning about "Version mismatch (beta-v0.1.6+t12). Please upgrade ckb-daemon"

closed 2015-08-05 by KraXarN

elewis787, 2015-08-05:
I will try the test branch when I can.Nice work with this and thank you for the quick response !