ccMSC/ckb #97: Ok I just found this Amazing project

badgero1234, 2015-08-05:
What Can I do for you regarding testing. Also on another related note; How difficult will it be to include the RGB Stafe model just announced a few hours ago, same keyboard Idea. What would need to change in the code? I am learning c and I figure if I have a project this will help me learn quicker. Thanks for your amazing work.

badgero1234, 2015-08-05:
also I am coding on a mac

mrueg, 2015-08-05:
See https://github.com/ccMSC/ckb/issues/94

badgero1234, 2015-08-11:
so how difficult will it be to add a new keyboard? like the strafe RGB?

ccMSC, 2015-08-12:
Depends entirely on how different the USB protocol is. It could be a simple matter of adding it to the supported devices, or it could require a complete rewrite. We won't know until it's released and someone gets the USB captures from the Windows driver.

(Closing this issue as a duplicate of #94 - feel free to continue the discussion there)

closed 2015-08-12 by ccMSC