ccMSC/ckb #98: Not recognizing/working after reboot until daemon restart

roboter202, 2015-08-05:
After system start/reboot the keyboard is not recognized by CKB and many keystrokes are missed.
The problem is solved after stoping and restarting the daemon or reinstalling.

NOTE: I have added the "usbhid.quirks=0x1B1C:0x1B13:0x4" parameter to grub

running Fedora 22 with Corsair K70 RGB

- besides the bug awesome program

osafi, 2015-08-05:
I'm having the same issue. I was previously on ckb beta-v0.1.3 with a keyboard firmware of 1.2 and everything was great. Started to have this problem when I updated to ckb beta-v0.1.6 and keyboard firmware 1.3

Running Fedora 22 with Corsair K95 RGB

ccMSC, 2015-08-06:
Try installing [the testing repo](https://github.com/ccMSC/ckb/tree/testing). It will most likely fix the problem, it's still a few days out from full release.

roboter202, 2015-08-06:
Indeed it did fix the problem. Thank you

fliiiix, 2015-08-06:
it's the same bug as #93

closed 2015-08-08 by ccMSC