ccMSC/ckb #102: Segmentation fault

zattack96, 2015-08-11:
I just tried to install ckb and whe i try to run it i get:

zacharch% ckb

(process:25150): Gtk-WARNING **: Locale not supported by C library.
	Using the fallback 'C' locale.
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
zsh: segmentation fault (core dumped)  ckb

Can anyone help?

fliiiix, 2015-08-11:
Adding the actual core dump data might help.

zattack96, 2015-08-11:
#0  0x0000000000436717 in ?? ()
#1  0x00000000004732ba in ?? ()
#2  0x00000000004d299d in ?? ()
#3  0x00007f16aaf4afea in QMetaObject::activate(QObject*, int, int, void**) () from /usr/lib/libQt5Core.so.5
#4  0x00007f16aaf57f38 in QTimer::timerEvent(QTimerEvent*) () from /usr/lib/libQt5Core.so.5
#5  0x00007f16aaf4be63 in QObject::event(QEvent*) () from /usr/lib/libQt5Core.so.5
#6  0x00007f16abfae00c in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from /usr/lib/libQt5Widgets.so.5
#7  0x00007f16abfb34e6 in QApplication::notify(QObject*, QEvent*) () from /usr/lib/libQt5Widgets.so.5
#8  0x00007f16aaf1c89b in QCoreApplication::notifyInternal(QObject*, QEvent*) () from /usr/lib/libQt5Core.so.5
#9  0x00007f16aaf7205d in QTimerInfoList::activateTimers() () from /usr/lib/libQt5Core.so.5
#10 0x00007f16aaf72561 in ?? () from /usr/lib/libQt5Core.so.5
#11 0x00007f16a99109fd in g_main_context_dispatch () from /usr/lib/libglib-2.0.so.0
#12 0x00007f16a9910ce0 in ?? () from /usr/lib/libglib-2.0.so.0
#13 0x00007f16a9910d8c in g_main_context_iteration () from /usr/lib/libglib-2.0.so.0
#14 0x00007f16aaf7323f in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/libQt5Core.so.5
#15 0x00007f16aaf1a26a in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from /usr/lib/libQt5Core.so.5
#16 0x00007f16aaf2220c in QCoreApplication::exec() () from /usr/lib/libQt5Core.so.5
#17 0x0000000000417730 in ?? ()
#18 0x00007f16a9e46790 in __libc_start_main () from /usr/lib/libc.so.6
#19 0x0000000000417e79 in ?? ()

ccMSC, 2015-08-12:
You need to compile a debug binary first. In the ckb directory use `./qmake-auto && make debug`

ccMSC, 2015-08-12:
Update: the problem is with the AUR package. It places the `ckb-animations` directory in `/opt/ckb` but then places `ckb` in `/usr/bin`, which means it isn't finding the animation scripts like it's supposed to.

In the testing repository I've fixed the crash that occurs, but it still won't have any animations. You need to contact the package maintainer.

Liganic, 2015-08-12:
AUR package maintainer here, fixed the package and verified that the crash is gone.

ccMSC, 2015-08-13:
Awesome. Will wait for @zattack96 to confirm before closing.

zattack96, 2015-08-13:
Crash is gone.

closed 2015-08-13 by zattack96